nextcloud/nextcloudpi #953: ncdata is not a BTRFS mountpoint.

Haraade, 2019-07-21:
nc-snapshot-auto stopped working after upgrade to Buster. When run nc-snapshot I get this message: ncdata is not a BTRFS mountpoint.

[image]
[image]

zedocrob, 2019-07-21:
Same issue for me

nachoparker, 2019-07-23:
I was able to replicate on my Odroid HC2. Thanks for reporting

nachoparker, 2019-07-25:
fixed. Thanks

closed 2019-07-25 by nachoparker

malbreitenbach, 2019-07-27:
Hey @nachoparker,
thanks for the fix and the great work, in the latest version the problem is solved on Raspi 4.