nextcloud/nextcloudpi #954: Installer script fails on Orange Pi PC+

jimgregory, 2019-07-21:
I have a fresh install version of Armbian 5.91 Debian Buster on an Orange Pi PC+.  I attempted to install using the autoinstaller in `armbian-config`, but it fails just after `jq` is installed.

I attempted to install using the installer script downloaded by `armbian-config`:
```
cd /tmp
wget https://github.com/nextcloud/nextcloudpi/archive/master/latest.tar.gz
tar -xzf latest.tar.gz
cd nextcloudpi-master
sudo bash install.sh
```
It  fails at `check_distro /etc.cfg`:
```
Installing NextCloudPi...
ERROR: distro not supported:
Debian Buster with Armbian Linux \l
```
The contents of /etc/issue are "`Debian Buster with Armbian Linux \l`" (that's a literal backslash and lower-case 'L').

Commenting out the distribution check  seems to resolve the problem.

igorpecovnik, 2019-07-21:
We recently changed the default issue so this problem doesn't show up on older images ... NCP installer is checking /etc/issue while it should rely on lsb_release

nachoparker, 2019-07-24:
thanks, good suggestion. It should work now.

closed 2019-07-24 by nachoparker