nextcloud/nextcloudpi #956: update to 16.0.3 fails with RedisException

violoncelloCH, 2019-07-21:
Maybe this is just not yet supported, but when trying to update to 16.0.3 via `ncp-update-nextcloud` I get this:

```
backup /var/www//nextcloud-bkp_20190721_1563738083.tar.gz generated
An unhandled exception has been thrown:
RedisException: MISCONF Redis is configured to save RDB snapshots, but it is currently not able to persist on disk. Commands that may modify the data set are disabled, because this instance is configured to report errors during writes if RDB snapshotting fails (stop-writes-on-bgsave-error option). Please check the Redis logs for details about the RDB error. in /var/www/nextcloud/lib/private/Memcache/Redis.php:64
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(64): Redis->setex('860ecd96bcb3550...', 60, '["\\/var\\/www\\/n...')
#1 /var/www/nextcloud/lib/autoloader.php(168): OC\Memcache\Redis->set('OC_User', Array, 60)
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/User/Session.php(223): spl_autoload_call('OC_User')
#4 /var/www/nextcloud/lib/private/L10N/Factory.php(168): OC\User\Session->getUser()
#5 /var/www/nextcloud/lib/private/L10N/Factory.php(127): OC\L10N\Factory->findLanguage('lib')
#6 /var/www/nextcloud/lib/private/Server.php(1507): OC\L10N\Factory->get('lib', NULL)
#7 /var/www/nextcloud/lib/base.php(234): OC\Server->getL10N('lib')
#8 /var/www/nextcloud/lib/base.php(645): OC::checkConfig()
#9 /var/www/nextcloud/lib/base.php(1056): OC::init()
#10 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#11 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#12 {main}Abort...
An unhandled exception has been thrown:
RedisException: MISCONF Redis is configured to save RDB snapshots, but it is currently not able to persist on disk. Commands that may modify the data set are disabled, because this instance is configured to report errors during writes if RDB snapshotting fails (stop-writes-on-bgsave-error option). Please check the Redis logs for details about the RDB error. in /var/www/nextcloud/lib/private/Memcache/Redis.php:64
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(64): Redis->setex('860ecd96bcb3550...', 60, '["\\/var\\/www\\/n...')
#1 /var/www/nextcloud/lib/autoloader.php(168): OC\Memcache\Redis->set('OC_User', Array, 60)
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/User/Session.php(223): spl_autoload_call('OC_User')
#4 /var/www/nextcloud/lib/private/L10N/Factory.php(168): OC\User\Session->getUser()
#5 /var/www/nextcloud/lib/private/L10N/Factory.php(127): OC\L10N\Factory->findLanguage('lib')
#6 /var/www/nextcloud/lib/private/Server.php(1507): OC\L10N\Factory->get('lib', NULL)
#7 /var/www/nextcloud/lib/base.php(234): OC\Server->getL10N('lib')
#8 /var/www/nextcloud/lib/base.php(645): OC::checkConfig()
#9 /var/www/nextcloud/lib/base.php(1056): OC::init()
#10 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#11 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#12 {main}Error backing up
Clean up...
Done. Press any key...
```

Also, it cleaned up the backup automatically, but did keep the NC maintenance mode on. Turning it off manually gives me back the working 16.0.2 setup luckily :)

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

grep: /etc/mysql/mariadb.conf.d/90-ncp.cnf: No such file or directory
NextCloudPi version  v1.16.0
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      386G/461G
rootfs usage         386G/461G
swapfile             /var/swap
dbdir                 (doesn't exist)
Nextcloud check      ok
Nextcloud version    16.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
[... 135 lines cut ...]
```

{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:27+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Cleanup invalid photocache files for carddav","userAgent":"--","version":"16.0.1.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:27+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add background job to cleanup login flow v2 tokens","userAgent":"--","version":"16.0.1.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:27+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove potentially over exposing share links","userAgent":"--","version":"16.0.1.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:27+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No need to remove link shares.","userAgent":"--","version":"16.0.1.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:27+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"16.0.1.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:30+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"16.0.1.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:30+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"16.0.2.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:30+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"16.0.2.1"}
{"reqId":"GQiKgAnJnfJbEZSdzAFp","level":1,"time":"2019-07-07T22:35:30+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"16.0.2.1"}
{"reqId":"ZpR5jh2iVzDoc8IvDQAG","level":3,"time":"2019-07-08T02:15:02+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"Undefined variable: response at \/var\/www\/nextcloud\/apps\/dav\/lib\/CardDAV\/SyncService.php#107","userAgent":"--","version":"16.0.2.1"}
{"reqId":"ZpR5jh2iVzDoc8IvDQAG","level":3,"time":"2019-07-08T02:15:02+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"Invalid argument supplied for foreach() at \/var\/www\/nextcloud\/apps\/dav\/lib\/CardDAV\/SyncService.php#107","userAgent":"--","version":"16.0.2.1"}
{"reqId":"ZpR5jh2iVzDoc8IvDQAG","level":3,"time":"2019-07-08T02:15:02+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"Undefined variable: response at \/var\/www\/nextcloud\/apps\/dav\/lib\/CardDAV\/SyncService.php#122","userAgent":"--","version":"16.0.2.1"}
{"reqId":"XSNkpn8AAQEAAHmeEj0AAAAH","level":3,"time":"2019-07-08T15:43:34+00:00","remoteAddr":"85.0.185.229","user":"--","app":"index","method":"GET","url":"\/apps\/files\/api\/v1\/showgridview","message":{"Exception":"TypeError","Message":"Argument 3 passed to OCA\\Files\\Service\\TagService::__construct() must be an instance of OCP\\ITags, null given, called in \/var\/www\/nextcloud\/apps\/files\/lib\/AppInfo\/Application.php on line 97","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/apps\/files\/lib\/AppInfo\/Application.php","line":97,"function":"__construct","class":"OCA\\Files\\Service\\TagService","type":"->","args":[{"__class__":"OC\\User\\Session"},{"__class__":"OC\\Activity\\Manager"},null,null,{"__class__":"Symfony\\Component\\EventDispatcher\\EventDispatcher"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OCA\\Files\\AppInfo\\{closure}","class":"OCA\\Files\\AppInfo\\Application","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["TagService"]},{"file [... cut ...] on":"16.0.2.1"}
{"reqId":"XTTBBjrkUESbB1meO@YM2QAAAAM","level":3,"time":"2019-07-21T19:46:14+00:00","remoteAddr":"89.217.131.2","user":"--","app":"PHP","method":"GET","url":"\/index.php\/204","message":"unlink(\/var\/www\/nextcloud\/data\/appdata_ocj2z7xiag7t\/css\/core\/c354-30fb-server.css.gzip): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#228","userAgent":"Mozilla\/5.0 (Android) Nextcloud-android\/3.7.0 RC3","version":"16.0.2.1"}

```
</details>

nachoparker, 2019-07-23:
Thanks for reporting. Looks like redis had some I/O problems? There is a ton of people that have run the upgrade command in the past year or two (including NC16.0.3) so I wonder what happened to you.

On a side note: even you are part of the nextcloud organization and have write access, please send your PRs against the `devel` branch so they can be reviewed before merge.

In your case it was a tiny typo so it's ok :)

violoncelloCH, 2019-07-23:
hi @nachoparker 
thanks for your response!
In the meantime ncp tried to do the update again (autoupdate) and failed awfully.
I guess because of no remaining diskspace (because of the backups?):
at least the data seems to be still there and the DB as well... (I don't know why but under /var/www/ncp-data.K648Fa)
but in /var/www/nextcloud/ there is an empty config.php (and a CAN_INSTALL)
I can't find the original config.php anywhere :/ any idea where I could retrieve it from? (I would have expected it in the backup archives, but can't find it there (only the apps/ and data/ directory seem to be there); also it includes all the old backup archives there as well, what I don't understand why (so multiplying diskspace usage with every backup))...

---
̣
I'm sorry about that... I was just too lazy to do a PR for a one character change :see_no_evil: will do a proper PR next time of course :)

nachoparker, 2019-08-24:
sounds like you need to free up some space before trying to upgrade?

if that's the case I'll close, but we can keep discussing and reopen if needed.

closed 2019-08-24 by nachoparker