nextcloud/nextcloudpi #957: NCP + Buster => imagick notification

Zakspeed, 2019-07-22:
after installing Booster in my NCP

"Overview"
```
Security & setup warnings
It's important for the security and performance of your instance that everything is correctly configured. To help you with that we are doing some automatic checks. Please see the linked documentation for more information.

There are some warnings regarding your setup.
This instance is missing some recommended PHP modules. For improved performance and better compatibility it is highly recommended to install them.
**imagick**
```

Anyone else has happened?

ovpc, 2019-07-22:
Yes, everyone has, imagick has been removed for security reasons.

Zakspeed, 2019-07-22:
> Yes, everyone has, imagick has been removed for security reasons.

Does this imply any performance problem in improving Nacho with the image gallery?

Co0p3r21, 2019-07-22:
It´s already discussed nextcloud/server#13099 and nextcloud/nextcloudpi#947

closed 2019-07-23 by nachoparker

nachoparker, 2019-07-23:
> Does this imply any performance problem in improving Nacho with the image gallery?

They paid zero attention to my proposal