nextcloud/nextcloudpi #959: Request Entity Too Large

musklor, 2019-07-25:
Hello, 
Nexctloudpi : v1.15.3
Error : Request Entity Too Large
When : upload

Post_max_size

!-- Paste the output of `sudo ncp-report` -->

musklor, 2019-07-25:
ModSecurity disabled, no more issue

nachoparker, 2019-07-25:
Thanks, yes this has been reported in the forums and I verified it.

Need to adjust modsecurity to Buster. Probably `SecRequestBodyLimit` again yes.

I don't know when I'll have time. Until then please disable modsecurity. PRs are welcome

nachoparker, 2019-07-25:
https://help.nextcloud.com/t/can-not-upload-files-larger-than-10mb-on-rpi4/57074/

musklor, 2019-07-26:
SecRequestBodyLimit tested without success

nachoparker, 2019-07-26:
thanks, last time we hopped distros the config files were layed out differently for modsecurity so there were adjustments to be made

musklor, 2019-07-27:
https://linuxtechlab.com/easy-way-to-integrate-apache-with-modsecurity-on-ubuntu/

musklor, 2019-07-27:
Maybe we need the V3 version ?

Ubuntu 15.04
Provided by @m2n and @akoul

libModSecurity
$ sudo apt-get install g++ flex bison curl doxygen libyajl-dev libgeoip-dev libtool dh-autoreconf libcurl4-gnutls-dev libxml2 libpcre++-dev libxml2-dev
$ cd /opt/
$ git clone https://github.com/SpiderLabs/ModSecurity
$ cd ModSecurity/
$ git checkout -b v3/master origin/v3/master
$ sh build.sh
$ git submodule init
$ git submodule update #[for bindings/python, others/libinjection, test/test-cases/secrules-language-tests]
$ ./configure
$ make
$ make install

Iolaum, 2019-09-07:
The upload setting for `SecRequestBodyNoFilesLimit` of 5MB as specified [here](https://github.com/nextcloud/nextcloudpi/blob/d5c1f0058b43c22ebb132f51df0ea144fbd927ec/bin/ncp/SECURITY/modsecurity.sh#L73-L74) looks suspiciously close to when the issue occurs.

I went ahead and tested changing it:

```
$ sudo -i
$ nano /etc/modsecurity/modsecurity_crs_99_whitelist.conf 
$ service apache2 restart
```

Made the following changes (included some artefacts of my testing)
```
# UPLOADS ( 5 MB max excluding file size )
#SecRequestBodyNoFilesLimit 5242880
SecRequestBodyNoFilesLimit 536870912
# Custom addition
SecRequestBodyLimit 536870912
```

- The changes to `SecRequestBodyLimit` didn't appear to have any effect!!
- Changing SecRequestBodyNoFilesLimit fixed the issue! (I 'm guessing up to the file size specified.)

According to [modesecurity documentation](https://github.com/SpiderLabs/ModSecurity/wiki/Reference-Manual-%28v2.x%29#secrequestbodynofileslimit) this should be about body size and not the whole file, maybe nextcloud-client sends the whole file in one body??

I can make a PR that raises the limit from 5mb to 512mb if people think it is a worthy workaround that doesn't add risk to users. At the moment I 've allowed it for my instance.

nachoparker, 2019-09-07:
> I can make a PR that raises the limit from 5mb to 512mb if people think it is a worthy workaround that doesn't add risk to users. At the moment I 've allowed it for my instance.

that would be great, thanks! I'll test your PR before merge.

In Stretch it was sufficient to touch `SecRequestBodyLimit`.

closed 2019-09-11 by nachoparker

7Elev3n, 2020-05-19:
Hi @nachoparker. For some reason, on my installation of NCP, if I change bodynofiles limit in modsecurity_crs_99_whitelist.conf (as done by modsecurity.sh), it has no effect on my uploading large files onto my cloud. However, if I change the same flag in modsecurity.conf, it fixes my uploading issue. Any idea why this might be happening? Are we supposed to change the former file, not the latter conf? Thank you!

unalignedcoder, 2023-01-13:
With Nextcloud 25, modSecurity 2.9.3 and CRS 3.2.2, this problem persists.

On the `REQUEST-903.9003-NEXTCLOUD-EXCLUSION-RULES.conf` file included in CRS there is a rule to change `SecRequestBodyLimit` at runtime, but it's not possible to do this with `SecRequestBodyNoFilesLimit`. (See [here](https://github.com/SpiderLabs/ModSecurity/issues/1933)).

Nextcloud however triggers the second limit and gets blocked by CRS when uploading, seemingly conflating data and file size either way.

gabooh, 2024-07-16:
Using :
- Nextcloud 29.0.2
- CRS 4.4.0
- nextcloud-rule-exclusions-plugin-1.2.0
- Apache 2.4.52

I had to change SecRequestBodyNoFilesLimit to 536870912 to fix the issue : 

```
<location "/remote.php/dav/files/">
     SecRequestBodyNoFilesLimit 536870912
  </location>
```