nextcloud/nextcloudpi #960: nc-automount replaces existing USB mountpoint with newly inserted

cloo, 2019-07-28:
After inserting a second drive (labeled "antras"), it was mounted at media/USBdrive, the same directory my primary USB drive was automatically mounted at. The primary drive with all my data was mounted at media/USBdrive1. 

Naturally, Owncloud stopped functioning until drive removed and rebooted. 
Tried several times exchanging USB slots - same result.

[image] 

**System information**
Impossible to copy all the content from trashy ncp-report produced, so here is the video (probably worth another bug report):
https://youtu.be/YZYaYJwlQzE

ovpc, 2019-07-28:
You should use [forum](https://help.nextcloud.com/tags/ncp) to get help.

closed 2019-07-28 by ovpc

cloo, 2019-07-28:
@ovpc I'm not for help here. I'm sure this is a bug.

> You should use [forum](https://help.nextcloud.com/tags/ncp) to get help.

cloo, 2019-07-28:
Ok, I agree I'm not the first to encounter this problem. But it should be stated somewhere that nc-automount will break your system if used for more than one drive without taking preparing drive labels.

ovpc, 2019-07-28:
Documentation and code for NCP are a work in progress, and open to everyone, to create, edit or to enhance, PM's to Development branch are most welcome. As is any assistance in editing and creating documentation. Join the wiki team at https://t.me/NCP_Wiki_Team for contributing to [docs.nextcloudpi.com](https://docs.nextcloudpi.com/en/how-to-configure-an-external-usb-drive-with-nextcloudpi/) Thank you !

ovpc, 2019-07-28:
If you read the [docs](https://docs.nextcloudpi.com/en/how-to-configure-an-external-usb-drive-with-nextcloudpi/), there is a warning about leaving /media/USBdrive when adding a second drive:

> Important note: If you are using only one USB drive, you can leave the default path /media/USBdrive for all configuration options, but if you are going to use more than one you need to reference them by label, or NCP will not be able to know which one is which. For instance, if the label of the drive is mylabel then use the path /media/mylabel for items such as nc-datadir, nc-database and so on.

cloo, 2019-07-28:
Really nice of you to care!
Lets try to understand why I made this mistake. Probably because I already saw instructions in nc-automount page. 

[image]

I didnt think that I need to look up something more - and there already was one warning, so it probably felt safe. 

Maybe you could expand that warning in nc-automount page, to look something like this: 

IMPORTANT:
1. halt or umount the drive before removing
2. before mounting more than one drive, make sure data locations are referenced by labels (not default). For more details see [docs](https://docs.nextcloudpi.com/en/how-to-configure-an-external-usb-drive-with-nextcloudpi/). 

And maybe the reasons why labels is not default, are not technical but historical? Maybe this could be changed for convenience?

ovpc, 2019-07-28:
I've added:

>  WARNING If you want add a second drive later, make sure you have added a LABEL to the first and all other drives to be mounted when formating, as they will then be mounted at /media/LABEL instead of /media/USBdrive

cloo, 2019-07-28:
I hope that will help the others. 
If I'll stay with nextcloud, I'll consider helping on documentation. 
Thanks.