nextcloud/nextcloudpi #961: After pulling new Docker container the nc-dir has changed

monsieurhannes, 2019-07-30:
Hi,
I wanted to upgrade to Buster, downloaded the recently updated Docker container, stopped my old container, pulled the new one and started it with the same parameters as the previous one.
The container starts, I can go in (bash) and `ncp-config` works as well, but the nextcloud is not reachable. 
I ran `ncp-update` and got this:
```
ncp-update
Downloading updates
Performing updates
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://security.debian.org/debian-security buster/updates InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
exfat-utils is already the newest version (1.3.0-1).
exfat-fuse is already the newest version (1.3.0-1).
php-smbclient is already the newest version (0.9.0-3+b1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
lsb-release is already the newest version (10.2019051400).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
NextCloudPi updated to version v1.15.3
```
So far so good, but it still doesn't work, fixing permissions got me this error:
```
Running nc-fix-permissions
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"OUC9h4lV91mp62DVNwle","level":3,"time":"2019-07-30T06:19:34+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/pubShare_737add2d.publicKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"OUC9h4lV91mp62DVNwle","level":3,"time":"2019-07-30T06:19:34+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/pubShare_737add2d.privateKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"OUC9h4lV91mp62DVNwle","level":3,"time":"2019-07-30T06:19:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/master_737add2d.publicKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"OUC9h4lV91mp62DVNwle","level":3,"time":"2019-07-30T06:19:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/master_737add2d.privateKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"OUC9h4lV91mp62DVNwle","level":3,"time":"2019-07-30T06:19:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/data\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.8.1"}
{"reqId":"OUC9h4lV91mp62DVNwle","level":3,"time":"2019-07-30T06:19:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at \/data\/nextcloud\/lib\/private\/Log\/File.php#141","userAgent":"--","version":"15.0.8.1"}
data directory not found
Done. Press any key...
```

Well, as I said before, I started it with the same parameters, `-v /media/data/nextcloud:/data`, which worked before. Now both (the stopped container) and the new have wrong location settings, it seems...

**Is that a known bug and is it easily fixable?** 

Last but not least the nc-info (the ports are supposed to be closed!):
```
Running nc-info
Gathering information...
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"Mp5SfaLVMvEflkCqPjB1","level":3,"time":"2019-07-30T06:22:08+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/pubShare_737add2d.publicKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"Mp5SfaLVMvEflkCqPjB1","level":3,"time":"2019-07-30T06:22:08+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/pubShare_737add2d.privateKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"Mp5SfaLVMvEflkCqPjB1","level":3,"time":"2019-07-30T06:22:13+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/master_737add2d.publicKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"Mp5SfaLVMvEflkCqPjB1","level":3,"time":"2019-07-30T06:22:14+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(\/media\/USBdrive\/ncdata\/files_encryption\/OC_DEFAULT_MODULE\/master_737add2d.privateKey): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"--","version":"15.0.8.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /data/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"Mp5SfaLVMvEflkCqPjB1","level":3,"time":"2019-07-30T06:22:14+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/data\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.8.1"}
{"reqId":"Mp5SfaLVMvEflkCqPjB1","level":3,"time":"2019-07-30T06:22:14+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at \/data\/nextcloud\/lib\/private\/Log\/File.php#141","userAgent":"--","version":"15.0.8.1"}
NextCloudPi version  v1.15.3
NextCloudPi image    NextCloudPi_docker_07-20-19
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          sda
datadir              /media/USBdrive/ncdata (doesn't exist)
rootfs usage         4.2G/59G
swapfile             none
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   172.17.0.2
gateway              172.17.0.1
interface            eth0
certificates         none
NAT loopback         no
uptime               35days
```

monsieurhannes, 2019-07-30:
When  I change the config.php to `/data/nextcloud/data` I get errors as well.
`ls /data/nextcloud/data` shows not what I'm used to see in other NCP I run:
`appdata_oc402fvgkjf4  files_encryption`

ovpc, 2019-07-30:
ports 80 and 443 should be open.
Did you remove the existing container? 

datadir should  be /data/nextcloud/data
So you could try and edit config.php to correct this. Edit: I see you tried that ;-)
but still weird! /media/USBdrive is a symlink created by nc-automount and not available in Docker image.

You could try creating a new container to a new empty datadir as in
> -v /media/data/nextcloud_new1:/data

monsieurhannes, 2019-07-30:
Ok, I remember the problem, but it's still strange:

I had everything running on NCP without docker on an external HDD. NCP created /media/USBdrive and mounted there. After switching to Docker I just kept that and everything was working without changing too much :P 
But now /media/USBdrive is not existing anymore.

Sorry, I know it wasn't the nicest way to handle the container...

monsieurhannes, 2019-07-30:
I can still see the files outside of Docker in `/media/data/files/ncdata`

ovpc, 2019-07-30:
Okay, will close this now, as clearly not a bug !

closed 2019-07-30 by ovpc

monsieurhannes, 2019-07-30:
Well, I'm not sure, when stopping the previously working docker container and downloading the new image, starting it and the data-dir doesn't work as before, I'd say it's not the way it should be...

ovpc, 2019-07-30:
You cant expect it to work with a datadir that does no longer exist!

If/when you delete all containers and start a new one pointing -v to /media/data/files/ncdata does it work or not? If that is where your files are.

monsieurhannes, 2019-07-30:
Well, this was my previous docker command, used it several times with the old docker container:
`sudo docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/data/nextcloud:/data -v /media/data/backup:/media/backup --name nextcloudpi_ag5 ownyourbits/nextcloudpi-armhf`

Now I used after finding the nc-data-dir:
`sudo docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/data/nextcloud:/data -v /media/data/backup:/media/backup -v /media/data/files/ncdata:/media/USBdrive --name nextcloudpi_ag7 ownyourbits/nextcloudpi-armhf`

I get no errors anymore, but a 503 Service Unavailable when I want to open NC.

monsieurhannes, 2019-07-30:
Do I have to delete the previous docker containers? Isn't it enough to stop them? I'm a bit afraid to delete them and won't be able to come back to a working NCP...

ovpc, 2019-07-30:
Yes, as mentioned existing containers, stopped or running, will block/reserve the assigned ports.
If dont want to delete, can use alternate ports for example
-p 4444:4443 -p 444:443 -p 81:80
But we should really take this and further questions to the [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

monsieurhannes, 2019-07-30:
[Started a thread in the forum,](https://help.nextcloud.com/t/nextcloud-datadir-changed-after-starting-the-new-container/57439) thank you for your help so far!

monsieurhannes, 2019-07-30:
A similar 503 error occured to another user:
https://help.nextcloud.com/t/503-service-unavailable-docker-image-ownyourbits-nextcloudpi-armh/57432