nextcloud/nextcloudpi #965: Missing PHP module in new Nextcloud Pi docker image

paschaef, 2019-08-07:
**System information**
 Nextcloud Pi Docker image `NextCloudPi_docker_07-20-19`
-  On Raspberry Pi (nextcloudpi-armhf with image ID: `3b832c3ddae7`)
- On x86 machine (nextcloudpi-x86 with image ID: `547600d6f319` )

**Problem**
After a fresh install of nextcloud pi in the above mention docker container, nextcloud reports a warning about a missing PHP module `imagick` in the administrators overview tab (see image). 
[image]

After installation the installation inside the docker container with
``` bash
apt-get install php-imagick
```
and a restart of the nextcloud container  the warning disappears. However, this change is not persistant in the container. After a complete shutdown of the container the effect of the above command has gone.

This php module should be already included in the docker image, so that no warning is generated in nextcloud admin page.

nero120, 2019-08-07:
@paschaef see [this](https://github.com/nextcloud/server/issues/13099) discussion in the nextcloud server repo, not so much a nextcloudpi issue.

paschaef, 2019-08-08:
@nero120 agree. As mentioned in https://github.com/nextcloud/nextcloudpi/pull/966#issuecomment-519026102 I wasn't aware that this is already an ongoing nextcloud issue.

closed 2019-10-08 by nachoparker