nextcloud/nextcloudpi #968: Modsecurity causes "connection closed" with large files

chase9, 2019-08-08:
**The Problem**
Over the weekend I moved my data to a flash drive connected to my pi. After setting up my instance again, my clients started throwing a "connection closed" error when trying to upload moderately sized files. My `/var/log/apache2/nc-error.log` was filled with these:

`ModSecurity: Request body no files data length is larger than the configured limit`

After disabling mod security from the NCP portal, my files are syncing again successfully. It appears mod security has a file size limit of about 512MB, which isn't affected by nc-limits. I propose that the mod security file size limits are either set with nc-limits, or a place to configure this limit is added to modsecurity in the NCP portal.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.15.3
NextCloudPi image    NextCloudPi_06-27-18
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.5G/29G
rootfs usage         3.5G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    16.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "63.226.182.197",
            "1": "192.168.1.50",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "4": "cloud.chaselau.me"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "16.0.3.0",
        "overwrite.cli.url": "https:\/\/cloud.chaselau.me\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
[... 96 lines cut ...]
```

{"reqId":"XUiCSB-EUzE49Gg6IuRfGQAAjQE","level":2,"time":"2019-08-05T19:23:52+00:00","remoteAddr":"73.94.133.55","user":"--","app":"core","method":"GET","url":"\/ocs\/v2.php\/cloud\/user?format=json","message":"Login failed: 'ncp' (Remote IP: '73.94.133.55')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.23.7","version":"16.0.3.0"}
{"reqId":"XUiCSB-EUzE49Gg6IuRfGwAAjQE","level":2,"time":"2019-08-05T19:23:53+00:00","remoteAddr":"73.94.133.55","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'ncp' (Remote IP: '73.94.133.55')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.23.7","version":"16.0.3.0"}
{"reqId":"XUsmI123x@wRYSCacZDfnQAAxQE","level":2,"time":"2019-08-07T19:27:31+00:00","remoteAddr":"73.94.133.55","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'ncp' (Remote IP: '73.94.133.55')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.23.7","version":"16.0.3.0"}
{"reqId":"XUsmI123x@wRYSCacZDfnAAAxQA","level":2,"time":"2019-08-07T19:27:32+00:00","remoteAddr":"73.94.133.55","user":"--","app":"core","method":"GET","url":"\/ocs\/v2.php\/cloud\/user?format=json","message":"Login failed: 'ncp' (Remote IP: '73.94.133.55')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.23.7","version":"16.0.3.0"}
{"reqId":"XUsmJF23x@wRYSCacZDfngAAwQA","level":2,"time":"2019-08-07T19:27:32+00:00","remoteAddr":"73.94.133.55","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'ncp' (Remote IP: '73.94.133.55')","userAgent":"Mozilla\/5.0 (iOS) Nextcloud-iOS\/2.23.7","version":"16.0.3.0"}
{"reqId":"XUxrLKmF5Cqkv845R6qnwAAAARU","level":3,"time":"2019-08-08T18:40:08+00:00","remoteAddr":"50.255.114.235","user":"chase","app":"no app in context","method":"PUT","url":"\/remote.php\/dav\/uploads\/chase\/552353658\/00000000","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 10000000 got 4702208","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Upload\/UploadFolder.php","line":45,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["00000000",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Upload\\UploadFolder","type":"->","args":["00000000",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["uploads\/chase\/552353658\/00000000",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/cloud.chaselau.me\/remote.php\/dav\/uploads\/chase\/552353658\/00000000","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpPut"],[{"absoluteUrl":"https:\/\/cloud.chaselau.me\/remote.php\/dav\/uploads\/chase\/552353658\/00000000","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:PUT",[{"absoluteUrl":"https:\/\/cloud.chaselau.me\/remote.php\/dav\/uploads\/chase\/552353658\/00000000","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/cloud.chaselau.me\/remote.php\/dav\/uploads\/chase\/552353658\/00000000","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Server.php","line":316,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":227,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.5.0daily (build 20181112) (Nextcloud)","version":"16.0.3.0"}
{"reqId":"XUxrLKmF5Cqkv845R6qnwQAAARY","level":3,"time":"2019-08-08T18:40:08+00:00","remoteAddr":"50.255.114.235","user":"chase","app":"no app in context","method":"PUT","url":"\/remote.php\/dav\/uploads\/chase\/2947913225\/00000000","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 10000000 got 3244032","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Upload\/UploadFolder.php","line":45,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["00000000",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Upload\\UploadFolder","type":"->","args":["00000000",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["uploads\/chase\/2947913225\/00000000",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/cloud.chaselau.me\/remote.php\/dav\/uploads\/chase\/29 [... cut ...] te.php","line":332,"function":"printPage","class":"OC\\Template\\Base","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/index.php","line":69,"function":"printExceptionErrorPage","class":"OC_Template","type":"::","args":[{"__class__":"RedisException"},500]}],"File":"\/var\/www\/nextcloud\/lib\/private\/Memcache\/Redis.php","Line":54,"CustomMessage":"--"},"userAgent":"fediverse.network crawler (https:\/\/fediverse.network\/about  root@fediverse.network)","version":"16.0.3.0"}

```
</details>

nachoparker, 2019-08-08:
Duplicate of https://github.com/nextcloud/nextcloudpi/issues/959

closed 2019-08-08 by nachoparker

kfirfer, 2020-08-23:
Tried to set this configurations in modsecurity-snippet on nginx ingress controller and getting connection closed when I trying to upload 700MB file
```
    nginx.ingress.kubernetes.io/modsecurity-snippet: |
      SecRuleEngine On
      SecRequestBodyAccess On
      SecResponseBodyAccess Off
      SecAuditEngine RelevantOnly
      SecAuditLogParts ABCIJDEFHZ
      SecAuditLog /var/log/modsec_audit.log
      SecRuleRemoveById 911100 941130 941170 932140 949110
      SecRequestBodyLimit 5368709120
      SecRequestBodyNoFilesLimit 5368709120
```