nextcloud/nextcloudpi #971: Internal Server Error with stack trace when trying to change password twice from forgot-password email

mihailstoynov, 2019-08-13:
When trying to change password twice from forgot-password email you get an Internal Server Error with stack trace that contains scary info

## How to reproduce
- Click on forgot password, request email
- Change password
- Go back to email and click on it again.
- You get a HTTP 500

## Result

```
Internal Server Error
The server was unable to complete your request.

If this happens again, please send the technical details below to the server administrator.

More details can be found in the server log.

Technical details
Remote Address: INTERNAL IP
Request ID: NMb5lwhzn5tPD1XYP81Y
Type: TypeError
Code: 0
Message: Argument 1 passed to OC\Security\Crypto::decrypt() must be of the type string, null given, called in /nextcloud_install_dir//core/Controller/LostController.php on line 189
File: /nextcloud_install_dir//lib/private/Security/Crypto.php
Line: 113

Trace
#0 /nextcloud_install_dir//core/Controller/LostController.php(189): OC\Security\Crypto->decrypt(NULL, 'user@gmail.comrf...')
#1 /nextcloud_install_dir//core/Controller/LostController.php(155): OC\Core\Controller\LostController->checkPasswordResetToken('kjo34HAK4...', 'USERNAME')
#2 /nextcloud_install_dir//lib/private/AppFramework/Http/Dispatcher.php(166): OC\Core\Controller\LostController->resetform('ji9397398...', 'username')
#3 /nextcloud_install_dir//lib/private/AppFramework/Http/Dispatcher.php(99): OC\AppFramework\Http\Dispatcher->executeController(Object(OC\Core\Controller\LostController), 'resetform')
#4 /nextcloud_install_dir//lib/private/AppFramework/App.php(126): OC\AppFramework\Http\Dispatcher->dispatch(Object(OC\Core\Controller\LostController), 'resetform')
#5 /nextcloud_install_dir//lib/private/AppFramework/Routing/RouteActionHandler.php(47): OC\AppFramework\App::main('OC\\Core\\Control...', 'resetform', Object(OC\AppFramework\DependencyInjection\DIContainer), Array)
#6 [internal function]: OC\AppFramework\Routing\RouteActionHandler->__invoke(Array)
#7 /nextcloud_install_dir//lib/private/Route/Router.php(297): call_user_func(Object(OC\AppFramework\Routing\RouteActionHandler), Array)
#8 /nextcloud_install_dir//lib/base.php(975): OC\Route\Router->match('/lostpassword/r...')
#9 /nextcloud_install_dir//index.php(42): OC::handleRequest()
#10 {main}
```

## Information disclosed
- username
- internal server ip ( that scared me a little)
- paths (/nextcloud_install_dir/)
- email
- part of the token

ovpc, 2019-08-13:
Pretty bad, but not a NextCloudPi issue, should be addressed to and by Nextcloud devs

mihailstoynov, 2019-08-13:
Oops, wrong project, sorry. You can close this one, I will resubmit it

closed 2019-08-13 by mihailstoynov