nextcloud/nextcloudpi #972: Reset password form for new users has an action that switches to http

mihailstoynov, 2019-08-13:
## The issue
When a new user is created a link is sent with a url to setup the initial password for that user.

The form inside this page looks like this:
`<form action="http://domain.com/lostpassword/set/lllHFuikfdAAAAAAu8Df/username" id="reset-password" method="post">
`
The problem is that the action contains the scheme and the domain name.

## Other forms don't have this issue

I checked other forms.
Login form:
`<form method="post" name="login">`
Forgot password form:
`<form method="post" name="login" action="lostpassword/email">`

They don't exibit this isuue.

## Possible solution

Can we make the change password form be like:
`<form action="/lostpassword/set/lllHFuikfdu8Df/username" id="reset-password" method="post">`
or
`<form action="../../../set/lllHFuikfdu8Df/username" id="reset-password" method="post">`

ovpc, 2019-08-13:
Again not NCP related,

closed 2019-08-13 by ovpc

mihailstoynov, 2019-08-13:
Sorry