nextcloud/nextcloudpi #975: custom code on auto-backup

OlafBr, 2019-08-20:
Hello,
when I want to execute some code (mount/unmount/compress/move/etc.) before or after the auto-backup, I have to edit the script /usr/local/bin/ncp-backup-auto. But this script is overwritten each time the configuration for the auto-backup is run: nc-backup-auto.sh. It is also overwritten by ncp updates (I suppose).

So I suggest to insert this lines into the script /usr/local/bin/ncp-backup-auto:
at the beginning:
	`[ -x /usr/local/bin/ncp-backup-auto-before ] && /usr/local/bin/ncp-backup-auto-before`
and at the end:
	`[ -x /usr/local/bin/ncp-backup-auto-after ] && /usr/local/bin/ncp-backup-auto-after`

So a user can create this two scripts, which are completely left alone by any configuration or updates. If the files do not exist, nothing happens.

For this enhancement the file bin/ncp/BACKUPS/nc-backup-auto.sh has to be modified:
Insert in line 21:
`[ -x /usr/local/bin/ncp-backup-auto-before ] && /usr/local/bin/ncp-backup-auto-before`
and insert in line 26 (before the "EOF":
`[ -x /usr/local/bin/ncp-backup-auto-after ] && /usr/local/bin/ncp-backup-auto-after`

Regards, Olaf

nachoparker, 2019-08-24:
Hi, 

Sounds good. Can you send a PR with this?

Thanks

OlafBr, 2019-08-26:
#978

closed 2019-09-02 by nachoparker