nextcloud/nextcloudpi #976: Debian buster Curl install

LsPjvVzszDt, 2019-08-21:
I've been trying to get the curl install run on a clen VM build of Debian Buster (from debian-10.0.0-amd64-netinst.iso).

The base build is minimal, no GUI but with curl, sudo and ssh.

Running the curl install has failed in a number of places. The first of which was due to /var/www/ncp-app not existing. After which I created /var/www/ncp-app and /var/www/ncp-web before running the installer.

It looks like it is getting very close to completing but is still failing, currently with:

> App "nextcloudpi" cannot be installed because appinfo file cannot be read.

Which is correct as  /var/www/nextcloud/apps/nextcloudpi is empty.

Pastebin of the install [here](https://pastebin.com/nMSfwWXf)

marco-aa, 2019-08-22:
Hello i was running into the same issue ( /var/www/ncp-app not present, same with ncp-web, i copied them by hand and did a chwon -R www-data:www-data and chmod 700 , chmod g+x on the folders, i had to do a userdel ncp for a aditional run of install.sh).

Finaly the script installed the ncp-web and i saw the Activation Page with the generated passwords.
Here i'm stuck ( i guess it's because of the before)

LsPjvVzszDt, 2019-08-22:
Thanks.

From a running server I can see that the missing directories would be set-up with:

mkdir -p /var/www/ncp-app/  /var/www/ncp-web/
chown root:root /var/www/ncp-app/
chmod 0755 /var/www/ncp-app/
chown www-data:www-data /var/www/ncp-web/
chmod 0770 /var/www/ncp-web/

However the curl install still ends at the same point with an empty nextcloudpi directoy.

uucwq, 2019-08-23:
Faced the same issue:  

> cp: cannot stat 'var/www/ncp-app' : No such file or directory

Solution posted by marco_aa helped to finish install.sh: 
I copied the folders ncp-app and ncp-web from Github into /var/www/ and set permissions manually. 

But now i'm stuck on the NextCloudPi Activation page (localhost/activate) with the following message after clicking on "Activate":

> NextCloudPi not yet initialized, trying again in a few seconds ...

chwon for ncp-config.d did not help. Any ideas?

marco-aa, 2019-08-23:
> chwon for ncp-config.d did not help. Any ideas?
i checked /var/log/apache2/error.log with a tail -f error.log and then i pressed "Activate"
Did you see errors ?

uucwq, 2019-08-23:
Okay I'm going to check the log after the weekend. 
But anyway how did you fix these put_content errors?

LsPjvVzszDt, 2019-08-23:
@nachoparker

Can we check if there are any pre-requisites that should be in place before running the curl installer?

Does this run successfully on Buster when you run it?

Thanks

LsPjvVzszDt, 2019-08-23:
**Workaround**

I used the following commands to get from a bare VM build to successful activation (YMMV):

> sudo -s
> 
> apt install unzip rsync
> 
> mkdir /root/temp/
> cd /root/temp/
> 
> wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
> chmod 0770 install.sh
> 
> 
> wget https://github.com/nextcloud/nextcloudpi/archive/master.zip
> unzip master.zip
> 
> mkdir -p /var/www/ncp-app/  /var/www/ncp-web/
> 
> chown root:root /var/www/ncp-app/
> chmod 0755 /var/www/ncp-app/
> 
> chown www-data:www-data /var/www/ncp-web/
> chmod 0770 /var/www/ncp-web/
> 
> rsync -avH nextcloudpi-master/ncp-app/ /var/www/ncp-app/
> rsync -avH nextcloudpi-master/ncp-web/ /var/www/ncp-web/
> 
> 
> mkdir -p /usr/local/etc/ncp-config.d/
> 
> chgrp -R www-data /usr/local/etc/ncp-config.d/
> chmod -R 2755     /usr/local/etc/ncp-config.d/
> 
> touch /usr/local/etc/ncp-config.d/nc-admin.cfg
> touch /usr/local/etc/ncp-config.d/nc-passwd.cfg 
> 
> chmod 0660  /usr/local/etc/ncp-config.d/nc-admin.cfg
> chmod 0660  /usr/local/etc/ncp-config.d/nc-passwd.cfg 
> 
> 
> ./install.sh

nachoparker, 2019-08-24:
one of the upstream dependencies changed, breaking the build. Fixed

closed 2019-08-24 by nachoparker

LsPjvVzszDt, 2019-08-25:
@nachoparker

This is all looking good now. Thank you.

One thing I've found is that **/usr/local/etc/ncp-version** is not created.