nextcloud/nextcloudpi #979: nc-rsync-auto.sh is missing port when doing ssh connection test

stratmaster, 2019-08-27:
In `nc-rsync-auto.sh` between from lines 31 to 35 the ssh connection is tested:

https://github.com/nextcloud/nextcloudpi/blob/d17fe03fb9162d15cc80b84b02b0edbc353d237f/bin/ncp/BACKUPS/nc-rsync-auto.sh#L31-L35

The problem is, that this test assumes port 22 by default. When a custom port is given, this connection test always fails with `SSH non-interactive not properly configured`. 

A solution might be to add the port in line 33 like this:

`local SSH=( ssh -o "BatchMode=yes" -p $PORTNUMBER "$NET" )`

Should I do a pull request?

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Nextcloud is in maintenance mode - no apps have been loaded

NextCloudPi version  v1.15.3
NextCloudPi image    NextCloudPi_07-21-19
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      156G/932G
rootfs usage         26G/29G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    16.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:28

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is in maintenance mode - no apps have been loaded

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "***REMOVED SENSITIVE VALUE***",
            "11": "***REMOVED SENSITIVE VALUE***",
            "4": "***REMOVED SENSITIVE VALUE***",
            "20": "***REMOVED SENSITIVE VALUE***",
            "21": "_",
            "22": "_",
            "3": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "16.0.3.0",
        "overwrite.cli.url": "***REMOVED SENSITIVE VALUE***",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/USBdrive\/ncdata\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "loglevel": "2",
        "log_type": "file",
        "maintenance": true,
        "logfile": "\/media\/USBdrive\/ncdata\/nextcloud.log",
        "htaccess.RewriteBase": "\/",
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtpport": "587",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "updater.secret": "***REMOVED SENSITIVE VALUE***"
    }
}

```
</details>

nachoparker, 2019-08-29:
> Should I do a pull request?

That would be awesome, thanks

Thanks for reporting.

nachoparker, 2019-09-07:
Thanks for reporting.

PR #983 (2019-09-02): Add missing port in nc-rsync-auto.sh
Closes #979
closed 2019-09-07 by nachoparker