nextcloud/nextcloudpi #989: Optimize database with mysqlcheck

feutl, 2019-09-04:
After running NC with NCP for more then a year with over 2 million files stored on it, I asked myself if the mysql/mariadb DB needs to be maintained?
I searched the internet and the simplest solution I found was using

```sudo mysqlcheck -o --all-databases```

This reduced the DB size from around 950MB to a little over 800MB and NC seems to be a little more responsive (but this could homeopathic as well)

Any, would it make sens to extend NCP with an "DB Optimize" option?

5 Days after the optimization plus an upgrade to NC 16.0.4 I am still under 850MB. I am fetching the information via ocs_data_server_database_size provided by serverinfo.

nachoparker, 2019-09-06:
not a bad idea!

md9931852945, 2019-09-18:
I would still suggest to add "Don't make assumptions on people's skills/abilities based on their outward appearance" as it would be a very good way to adress the underlying issue in my opinion.
ocs_data_server_database_size provided by serverinfo.

schoetju, 2023-11-25:
I've added this feature to NCP (see above pull request 1852 ).
Question: Should this feature be added to happen automatically? Like db-optimize-auto triggered e.g. on a weekly or monthly basis?

theCalcaholic, 2023-11-29:
@schoetju Imo, it should be automatic. If I understand it right, we don't even need to expose it in the admin interface - a weekly Cronjob would be enough.

This means, I'd probably integrate it into lamp.sh 🤔

schoetju, 2023-11-29:
I see it the same way. It doesn't need the admin interface (however - it's been easier for me to work it out like this). 
Instead it should be added as a weekly CRON job directly during the new installation or during the NCP version update. 
@theCalcaholic Happy, if you can move on with integrating it.