nextcloud/nextcloudpi #994: Hardware support request: any/all performant Olimex SBC

jcstaudt, 2019-09-08:
**Description**  
I don't represent Olimex, but I believe it would be highly valuable to enable NextcloudPi on any/all performant Olimex OLinuXino-series boards as they are [certified OSHW](https://certification.oshwa.org/list.html). I believe this is a primary motivation for FreedomBox choosing the A20-OLinuXino-LIME2 for [their platform](https://www.olimex.com/Products/OLinuXino/Home-Server/Pioneer-FreedomBox-HSK) and, as a result, receiving much praise from the free software community.

**Example**  
Add support for any performant Olimex OLinuXino board and release images for:

- [A64-OLinuXino](https://www.olimex.com/Products/OLinuXino/A64/A64-OLinuXino/open-source-hardware) (quad-core; 1GbE; USB2.0)
- [A20-OLinuXino-LIME2](https://www.olimex.com/Products/OLinuXino/A20/A20-OLinuXino-LIME2/open-source-hardware) (dual-core; 1GbE; SATA; [2x uSD](https://www.olimex.com/Products/OLinuXino/A20/Lime2-SD/open-source-hardware))
- others?

nachoparker, 2019-09-10:
We can add support if

1) it is supported by Armbian
2) there is a decent userbase

jcstaudt, 2019-09-10:
Yes, the boards are [supported by Armbian](https://www.armbian.com/download/?tx_maker=olimex). I have no idea how prevalent the boards are, but there is an active community within the [Olimex](https://www.olimex.com/forum/index.php) and [FreedomBox](https://discuss.freedombox.org/) forums at a minimum and I would argue that there is a decent and enthusiastic userbase. Tsvetan at Olimex would have to advise regarding the numbers.

testbird, 2020-04-20:
Hello @nachoparker as we were looking at the freedombox group effort a while ago, I looked again, and found some interesting things, they added support for more boards, implemented NAS features in master, and have nextcloud integration up on the their [Roadmap2020](https://wiki.debian.org/FreedomBox/Roadmap2020).