nextcloud/nextcloudpi #995: updating from 1.16.0. to 1.16.0 since 3 days now

JimmyKater, 2019-09-09:
i tried to update to v1.16.1 manually and got this:

> $ sudo ncp-update
> Downloading updates
> Performing updates
> Update to buster available. Type 'sudo ncp-dist-upgrade' to upgrade
> Cannot load Zend OPcache - it was already loaded
> Unknown user

and that's it. if i'd try to run dist-upgrade i get warned that it's not yet needed so i declined it.
since then automatic nc-update (which is enabled) tries to update itself and i always will get a message, that it was updated to 1.16.0... since 3 days now.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.16.0
NextCloudPi image    NextCloudPi_01-14-19
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      68G/916G
rootfs usage         2,5G/3,7G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    16.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               18days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "#1",
            "2": "#2",
            "4": "#3",
            "5": "#4",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
[... 102 lines cut ...]
```

{"reqId":"T3rjMncnIktIXmRap6CB","level":1,"time":"2019-09-09T08:45:15+00:00","remoteAddr":"","user":"--","app":"social","method":"","url":"--","message":"[>>] request: {\"protocol\":\"https\",\"host\":\"mastodon.technology\",\"url\":\"\\\/users\\\/ctonysem\",\"timeout\":10,\"type\":0,\"data\":[]} - content-type: application\/activity+json; charset=utf-8 - result: {\"@context\":[\"https:\/\/www.w3.org\/ns\/activitystreams\",\"https:\/\/w3id.org\/security\/v1\",{\"manuallyApprovesFollowers\":\"as:manuallyApprovesFollowers\",\"toot\":\"http:\/\/joinmastodon.org\/ns#\",\"featured\":{\"@id\":\"toot:featured\",\"@type\":\"@id\"},\"alsoKnownAs\":{\"@id\":\"as:alsoKnownAs\",\"@type\":\"@id\"},\"movedTo\":{\"@id\":\"as:movedTo\",\"@type\":\"@id\"},\"schema\":\"http:\/\/schema.org#\",\"PropertyValue\":\"schema:PropertyValue\",\"value\":\"schema:value\",\"Hashtag\":\"as:Hashtag\",\"Emoji\":\"toot:Emoji\",\"IdentityProof\":\"toot:IdentityProof\"}],\"id\":\"https:\/\/mastodon.technology\/users\/ctonysem\",\"type\":\"Person\",\"following\":\"https:\/\/mastodon.technology\/users\/ctonysem\/following\",\"followers\":\"https:\/\/mastodon.technology\/users\/ctonysem\/followers\",\"inbox\":\"https:\/\/mastodon.technology\/users\/ctonysem\/inbox\",\"outbox\":\"https:\/\/mastodon.technology\/users\/ctonysem\/outbox\",\"featured\":\"https:\/\/mastodon.technology\/users\/ctonysem\/collections\/featured\",\"preferredUsername\":\"ctonysem\",\"name\":\"\",\"summary\":\"\\u003cp\\u003e\\u003c\/p\\u003e\",\"url\":\"https:\/\/mastodon.technology\/@ctonysem\",\"manuallyApprovesFollowers\":false,\"publicKey\":{\"id\":\"https:\/\/mastodon.technology\/users\/ctonysem#main-key\",\"owner\":\"https:\/\/mastodon.technology\/users\/ctonysem\",\"publicKeyPem\":\"-----BEGIN PUBLIC KEY-----\\nMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAvNM1YeelRGUI1hrU3jlp\\nsJjJ4gZujFNXAFo8asle8qC22wr1qLdrAOhO3MKY0M3hnaeiCEc4UUrQa3CTAgut\\nbMCNUk\/LqykGkkqn706HJgwTUUZi3AhbSrZD7roIhbPm7ZtC\/TrQYfB0bSCO2t\/3\\n+ver2Nfng7PwAfmhg\/FLWIBdOQsjvchH41U+DsKxZSlKCVKAAaYBSAZZOcDrrJTh\\nwU30U0vhf8nJfNwRjXEHyggTCax6UFJPIB1n3EyXCQhN8MFa60HUONDSqmapNftk\\nYYPva8WixRUDfpXXDtv4R+w5Q4RARuwy8+WIToGGg2eJnYdIo2p2KBfH4Hxiw\/ij\\nqQIDAQAB\\n-----END PUBLIC KEY-----\\n\"},\"tag\":[],\"attachment\":[],\"endpoints\":{\"sharedInbox\":\"https:\/\/mastodon.technology\/inbox\"}}","userAgent":"--","version":"16.0.2.1"}
{"reqId":"T3rjMncnIktIXmRap6CB","level":1,"time":"2019-09-09T08:45:16+00:00","remoteAddr":"","user":"--","app":"social","method":"","url":"--","message":"[>>] request: {\"protocol\":\"https\",\"host\":\"cloud.pquirk.com\",\"url\":\"\\\/index.php\\\/apps\\\/social\\\/@Paul\",\"timeout\":10,\"type\":0,\"data\":[]} - content-type: application\/json; charset=utf-8 - result: {\"@context\":[\"https:\\\/\\\/www.w3.org\\\/ns\\\/activitystreams\",\"https:\\\/\\\/w3id.org\\\/security\\\/v1\"],\"id\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\",\"type\":\"Person\",\"url\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\",\"local\":true,\"aliases\":[\"@Paul\",\"users\\\/Paul\"],\"preferredUsername\":\"Paul\",\"name\":\"\",\"inbox\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\\\/inbox\",\"outbox\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\\\/outbox\",\"account\":\"Paul@cloud.pquirk.com\",\"following\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\\\/following\",\"followers\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\\\/followers\",\"endpoints\":{\"sharedInbox\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/inbox\"},\"publicKey\":{\"id\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul#main-key\",\"owner\":\"https:\\\/\\\/cloud.pquirk.com\\\/index.php\\\/apps\\\/social\\\/@Paul\",\"publicKeyPem\":\"-----BEGIN PUBLIC KEY-----\\nMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEA1slOzo\\\/nymF2rgfN72iw\\nJP\\\/3VrHrk2FWgKJpM2HXa6FQzhNU8zxoik4H8Z4pdhqfu2X5AalhSotbT28R1zfO\\n9wGS+6IXtkm3S5fiurnuc8MNpIzDwRYLbqOYMWq4wt+UDGdazuiGWh7MGxFYq0EU\\njgma4tTdSf0Gqi7ZSDe0TWWAojyrnUDrOKcYZxFVJJV9E4PwfvnFF2O8KmyHH4zq\\n7m6VIEn3IxTrEzRt1L8uCBKnZ\\\/MJminE2pGW4nrrSYHipyoPO09Uu9o1KvMh\\\/z59\\n0n73TwUt0COdgg4zkbLanyrS\\\/8mwBjPyxROPiVJDIsiYSRgmUB1vlQL\\\/yNqX7a1T\\nhwIDAQAB\\n-----END PUBLIC KEY-----\\n\"}}","userAgent":"--","version":"16.0.2.1"}
{"reqId":"T3rjMncnIktIXmRap6CB","level":1,"time":"2019-09-09T08:45:16+00:00","remoteAddr":"","user":"--","app":"social","method":"","url":"--","message":"[>>] request: {\"protocol\":\"https\",\"host\":\"social.tchncs.de\",\"url\":\"\\\/users\\\/homer77\",\"timeout\":10,\"type\":0,\"data\":[]} - content-type: application\/activity+json; charset=utf-8 - result: {\"@context\":[\"https:\/\/www.w3.org\/ns\/activitystreams\",\"https:\/\/w3id.org\/security\/v1\",{\"manuallyApprovesFollowers\":\"as:manuallyApprovesFollowers\",\"toot\":\"http:\/\/joinmastodon.org\/ns#\",\"featured\":{\"@id\":\"toot:featured\",\"@type\":\"@id\"},\"alsoKnownAs\":{\"@id\":\"as:alsoKnownAs\",\"@type\":\"@id\"},\"movedTo\":{\"@id\":\"as:movedTo\",\"@type\":\"@id\"},\"schema\":\"http:\/\/schema.org#\",\"PropertyValue\":\"schema:PropertyValue\",\"value\":\"schema:value\",\"Hashtag\":\"as:Hashtag\",\"Emoji\":\"toot:Emoji\",\"IdentityProof\":\"toot:IdentityProof\",\"focalPoint\":{\"@container\":\"@list\",\"@id\":\"toot:focalPoint\"}}],\"id\":\"https:\/\/social.tchncs.de\/users\/homer77\",\"type\":\"Person\",\"following\":\"https:\/\/social.tchncs.de\/users\/homer77\/following\",\"followers\":\"https:\/\/social.tchncs.de\/users\/homer77\/followers\",\"inbox\":\"https:\/\/social.tchncs.de\/users\/homer77\/inbox\",\"outbox\":\"https:\/\/social.tchncs.de\/users\/homer77\/outbox\",\"featured\":\"https:\/\/social.tchncs.de\/users\/homer77\/collections\/featured\",\"preferredUsername\":\"homer77\",\"name\":\"Homer S.\",\"summary\":\"\\u003cp\\u003e:xmpp: homer77@blabber.im\\u003cbr \/\\u003e:matrix: \\u003cspan class=\\\"h-card\\\"\\u003e\\u003ca href=\\\"https:\/\/social.tchncs.de\/@homer77\\\" class=\\\"u-url mention\\\"\\u003e@\\u003cspan\\u003ehomer77\\u003c\/span\\u003e\\u003c\/a\\u003e\\u003c\/span\\u003e:matrix.org\\u003cbr \/\\u003eactivist from \\u003ca href=\\\"https:\/\/social.tchncs.de\/tags\/muenster\\\" class=\\\"mention hashtag\\\" rel=\\\"tag\\\"\\u003e#\\u003cspan\\u003emuenster\\u003c\/span\\u003e\\u003c\/a\\u003e, Germany\\u003cbr \/\\u [... cut ...]
{"reqId":"2LSj5Bh1oWA3ozBHdChM","level":1,"time":"2019-09-09T08:50:20+00:00","remoteAddr":"","user":"--","app":"passwords","method":"","url":"--","message":"Passwords runs \/var\/www\/nextcloud\/occ in global mode","userAgent":"--","version":"16.0.2.1"}
{"reqId":"fxEVHLoz6xBBzrdcS7Zk","level":1,"time":"2019-09-09T08:50:26+00:00","remoteAddr":"","user":"--","app":"passwords","method":"","url":"--","message":"Passwords runs \/var\/www\/nextcloud\/occ in global mode","userAgent":"--","version":"16.0.2.1"}

```
</details>

nachoparker, 2019-09-10:
>  Update to buster available. Type 'sudo ncp-dist-upgrade' to upgrade

You can safely ignore this if you don't want to do the dist upgrade yet

closed 2019-09-10 by nachoparker

JimmyKater, 2019-09-10:
> You can safely ignore this if you don't want to do the dist upgrade yet

that's cool. i did so.
but as a matter of fact this wasn't my problem. my problem was that ncp was updating to the very same version. **from v1.16.0 to v1.16.0. for 3 days in a row.** 

that solved itself tonight by updating automatically to v1.17.0

nachoparker, 2019-09-11:
oh I see. Thanks for the heads up, I'll keep an eye on this :)