nextcloud/nextcloudpi #996: Maintenance mode

LsPjvVzszDt, 2019-09-10:
NextCloudPi version  v1.16.3

If maintenance mode is enabled nc-backup or nc-restore disable it, rather than leave it in the state it was prior to running.

rababerladuseladim, 2019-11-10:
What would be a good way to deal with this? 
My first idea: script queries the maintenance status at start and stores bool in variable MAINT_MODE_ACTIVATED_BY_SELF. Once the script is done, prior to exit, maint mode is disable only if MAINT_MODE_ACTIVATED_BY_SELF.

LsPjvVzszDt, 2019-11-10:
Any module that can affect maintenance mode checks for the current maintenance mode on startup. Performs its task. Resets the needed maintenance mode on exit. Sometimes this may mean do nothing on exit.

In my case I had set maintenance mode on using nc-maintenance prior to using nc-backup.

It could be as simple as 

if [ -f  /usr/local/etc/ncp-config.d/nc-maintenance-mode-on ]; then...

rababerladuseladim, 2019-11-11:
I like the idea very much. A drawback of the file-based check is in my opinion, that the status of maintenance mode (*MM*) will only be respected if it was initiated by a ncp script. If I turn on MM using the occ command, this file would not be written and ncp-scripts would therefore turn off MM on exit. 
To me, checking the MM on startup seems therefore like a more robust approach. 

However, I can imagine hiccups resulting from either implementation:
Imagine script 1 activates MM and does it's thing. While script 1 is still running, script 2 is started and takes a note: "I shall not turn off MM when I'm done!". While script 2 does it's job, script 1 finishes and either (a) turns off, or (b) checks somehow if another script requires MM. In case b, script 2 now needs to know somehow, that it is the only remaining script requiring MM. Otherwise, MM will stay on forever. :-(

This doesn't sound easy to fix...

LsPjvVzszDt, 2019-11-11:
Then how about using the occ/ncc command ?

if [ $(ncc maintenance:mode | grep -c1 "disabled" ) ]
then
	echo "Not in maintenance mode"
else
	echo "In maintenance mode"
fi

nachoparker, 2019-11-13:
Or

Every ncp-app that goes into maintenance mode adds a counter and every ncp-app that ends retrieves a counter. If a ncp-app retrieves the last counter then turn maintenance mode off.

LsPjvVzszDt, 2019-11-14:
Would this account for MM being set outside of ncp?
Do the ncp apps that change MM call occ/ncc when they want to change MM?

nachoparker, 2019-11-16:
> Would this account for MM being set outside of ncp?
no, but it could be taken into account (start the counter with 1 if it is already in MM)

> Do the ncp apps that change MM call occ/ncc when they want to change MM?
yes