nextcloud/nextcloudpi #1000: [ nc-datadir ] possible bug with USBdevice, nc-automat, nc-format-USB

Multisaft, 2019-09-13:
Regarding this topic from nextcloud forum:
https://help.nextcloud.com/t/ncp-cannot-move-datadir/59845/6

is it possible that there is an Issue with the nc-datadir script? does it point not to the device but to the link (myCloudDrive vs. USBdrive).

pasted from the original thread:
OK,
i started from scratch, with 
1. buster lite,
2. apt-get update/upgrade
3. password change of pi user
4. sudo bash
5. raspberrypi installation script ```
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
6. after setup, wizard start for change of data directory to usb storage:
    [ nc-automount ]
    Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount.service → /usr/lib/systemd/system/nc-automount.service.
    Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service → /usr/lib/systemd/system/nc-automount-links.service.
    automount enabled

    [ nc-format-USB ]
    /dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
    /dev/sda: 8 bytes were erased at offset 0xdf90355e00 (gpt): 45 46 49 20 50 41 52 54
    /dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
    Drive sda formatted successfuly and labeled myCloudDrive

    [ nc-datadir ]
    Maintenance mode enabled
    moving data directory from /var/www/nextcloud/data to /media/USBdrive/ncdata...
    BTRFS filesystem detected
    ERROR: target path already exists: /media/myCloudDrive/ncdata

ls -l /media gives me
total 16

drwxr-xr-x 1 root root 0 Sep 13 12:29 myCloudDrive
lrwxrwxrwx 1 root root 19 Sep 13 12:29 USBdrive -> /media/myCloudDrive

so Link is /media/USBdrive
to Device /media/myCloudDrive
as it was formated to btrfs from [ nc-format-USB ]

the chmod looks good right? read write access for root

if found that discussion on githup from 2017 (quite old I know):
https://github.com/nextcloud/nextcloudpi/issues/305

there someone has the same strange behaviour
I've tried almost everything from nachoparker with removed media symlinks,
changed devicenames
and especially wayneouttheres description of his fix: https://github.com/nextcloud/nextcloudpi/issues/305#issuecomment-377697798

but when I switch to ncp control panel to use nc-datadir with the Data directory:
/media/myCloudDrive/ncdata

it gives:
[ nc-datadir ]
Nextcloud is in maintenance mode - no apps have been loaded

INFO: /media/myCloudDrive/ncdata is not empty. Creating backup /media/myCloudDrive/ncdata-09-13-19
Maintenance mode already enabled
moving data directory from /var/www/nextcloud/data to /media/myCloudDrive/ncdata...
BTRFS filesystem detected
ERROR: target path already exists: /media/myCloudDrive/ncdata

7. the actual status of ncp-report gives:

> <--! Paste this in GitHub report -->

<details>
[... 134 lines cut ...]
2019-09-13 12:27:46 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2019-09-13 12:27:46 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2019-09-13 12:27:46 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2019-09-13 12:27:46 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2019-09-13 12:27:46 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2019-09-13 12:27:46 0 [Note] InnoDB: 10.3.17 started; log sequence number 6801485; transaction id 5149
2019-09-13 12:27:46 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2019-09-13 12:27:46 0 [Note] Plugin 'FEEDBACK' is disabled.
2019-09-13 12:27:46 0 [Note] Server socket created on IP: '127.0.0.1'.
2019-09-13 12:27:46 0 [Note] Reading of all Master_info entries succeeded
2019-09-13 12:27:46 0 [Note] Added new Master_info '' to hash table
2019-09-13 12:27:46 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.17-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
2019-09-13 12:27:47 0 [Note] InnoDB: Buffer pool(s) load completed at 190913 12:27:47

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

FuerteLoco, 2019-09-13:
I can confirm, that there is something wrong (see [nc-datadir.sh](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-datadir.sh)): `DATADIR` is always created and it is used to determine the "real" path name (instead of the symlink). But if you are using BTRFS, then the subvolume cannot be created, since it is the same as the directory created before:

```bash
  # resolve symlinks and use the real path
  mkdir -p "$DATADIR"
  DATADIR=$(cd "$DATADIR" && pwd -P)

  # use subvolumes, if BTRFS
  [[ "$( stat -fc%T "$BASEDIR" )" == "btrfs" ]] && {
    echo "BTRFS filesystem detected"
    btrfs subvolume create "$DATADIR" || return  1
  }
```
I made a temporary workaround and inserted `rm -r "$DATADIR"` before the subvolume creation. Now it started copying my files. Fingers crossed that nothing goes wrong.

JDC8077, 2019-09-14:
I am having the same, using last NCP OdroidHC2 image, updated to 1.16.5.

Currently blocked on new install, Linux skills weak so uncertain about how to apply the workaround above.

nachoparker, 2019-09-14:
I pushed a fix in `v1.16.6`. Please confirm that it works for you

closed 2019-09-14 by nachoparker

JDC8077, 2019-09-15:
Many Thanks 

> 
> 
> I pushed a fix in `v1.16.6`. Please confirm that it works for you