nextcloud/nextcloudpi #1003: Appstore Not available

rspierenburg, 2019-09-16:
I'm having an issue with NextcloudPi and retrieving the app store.  I have NextcloudPI running on an Odroid HC-1 with a second USB harddrive for Backups attached.  I ran NCP-Report (see below) and from what I can gather it is trying to access /media/USBdrive/tmp which it says it can't.  Well thats because /media/USBdrive is my backup drive and /media/USBdrive1 is my ncdata drive.  I have /media/myCloudDrive/ncdata set in the WebUI which is a symbolic link to USBdrive1.

See report:
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.16.6
NextCloudPi image    NextCloudPi_08-02-19
distribution         Debian Buster with Armbian Linux \l
automount            yes
USB devices          sda sdb
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      371G/3.7T
rootfs usage         3.8G/7.2G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    16.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0630ca1a
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               12min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "45.2.197.30",
            "1": "192.168.1.9",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "20": "spierenburg.ca",
            "21": "www.spierenburg.ca",
            "22": "192.168.1.9",
            "4": "www.spierenburg.ca"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "16.0.4.1",
        "overwrite.cli.url": "https:\/\/www.spierenburg.ca\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
[... 92 lines cut ...]
```

{"reqId":"XX@RgFi8RXahnsT2sQePQQAAAAI","level":2,"time":"2019-09-16T13:43:28+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@RgFi8RXahnsT2sQePQQAAAAI","level":2,"time":"2019-09-16T13:43:28+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@RgFi8RXahnsT2sQePQQAAAAI","level":2,"time":"2019-09-16T13:43:28+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@RnaZHJuLzMBR7@0uJ1gAAAEs","level":2,"time":"2019-09-16T13:43:57+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@RnaZHJuLzMBR7@0uJ1gAAAEs","level":2,"time":"2019-09-16T13:43:57+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@RnaZHJuLzMBR7@0uJ1gAAAEs","level":2,"time":"2019-09-16T13:43:57+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@Ru6ZHJuLzMBR7@0uJ1wAAAE0","level":2,"time":"2019-09-16T13:44:28+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@Ru6ZHJuLzMBR7@0uJ1wAAAE0","level":2,"time":"2019-09-16T13:44:28+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@Ru6ZHJuLzMBR7@0uJ1wAAAE0","level":2,"time":"2019-09-16T13:44:28+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"ABblfpS9TVtf42Davr2w","level":2,"time":"2019-09-16T13:44:54+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"--","version":"16.0.4.1"}
{"reqId":"XX@R2KZHJuLzMBR7@0uJ2AAAAFA","level":2,"time":"2019-09-16T13:44:56+00:00","remoteAddr":"142.177.220.201","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Jakarta Commons-HttpClient\/3.1","version":"16.0.4.1"}
{"reqId":"XX@R2KZHJuLzMBR7@0uJ2AAAAFA","level":2,"time":"2019-09-16T13:44:56+00:00","remoteAddr":"142.177.220.201","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Jakarta Commons-HttpClient\/3.1","version":"16.0.4.1"}
{"reqId":"XX@R2KZHJuLzMBR7@0uJ2AAAAFA","level":2,"time":"2019-09-16T13:44:56+00:00","remoteAddr":"142.177.220.201","user":"--","app":"no app in context","method":"GET","url":"\/index.php\/204","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Jakarta Commons-HttpClient\/3.1","version":"16.0.4.1"}
{"reqId":"XX@R2Fi8RXahnsT2sQePQgAAAAQ","level":2,"time":"2019-09-16T13:44:56+00:00","remoteAddr":"142.177.220.201","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) Nextcloud-android\/3.8.0","version":"16.0.4.1"}
{"reqId":"XX@R2Fi8RXahnsT2sQePQgAAAAQ","level":2,"time":"2019-09-16T13:44:56+00:00","remoteAddr":"142.177.220.201","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Android) Nextcloud-android\/3.8.0","version":"16.0.4.1"}
{"reqId":"XX@R2Fi8RXahnsT2sQePQgAAAAQ","level":2,"time":"2019-09-16T13:44:56+00:00","remoteAddr":"142.177.220.201","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozi [... cut ...] illa\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}
{"reqId":"XX@R2qZHJuLzMBR7@0uJ2QAAAFM","level":2,"time":"2019-09-16T13:44:58+00:00","remoteAddr":"192.168.1.1","user":"robert","app":"no app in context","method":"PROPFIND","url":"\/remote.php\/dav\/files\/robert\/Music","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"16.0.4.1"}

```
</details>

ovpc, 2019-09-16:
Go to forum please, this is not for [support](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm).
in config.php 
for 
tempdirectory and logfile change
USBdrive
into
myCloudDrive

closed 2019-09-16 by ovpc

rspierenburg, 2019-09-16:
Thank you for the information, I apologize for not posting in the forums.  However, I still believe this IS an issue.  Shouldn't NextcloudPI update the "tempdirectory" to match the ncdata location automatically?