nextcloud/nextcloudpi #1004: ncp-restore: sed: can't read /etc/php/7.2/mods-available/opcache.ini

feutl, 2019-09-17:
I was running ncp-restore yesterday to restore my database and at the end this error occurred
```sed: can't read /etc/php/7.2/mods-available/opcache.ini: No such file or directory```

Because I am running Debian 10 with 1.16.6 NCP I am quite sure this is just a mistake not setting 7.3 PHP instead of 7.2 :)
Btw. the backup was made a day before on the same machine already running Debian 10 and NCP in the 1.16.6 version.
The backup gets created daily with the autobackup feature.

The rest worked just fine, I was more than surprised and happy because my calendar was back :D so thanks.

```System Information
NextCloudPi version  v1.16.6
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /media/nextcloud-data/
data in SD           no
data filesystem      nfs
data disk usage      5.5T/7.1T
rootfs usage         12G/225G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    16.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       no
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp9s0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               7days

```

nachoparker, 2019-09-17:
that's weird. What are the contents of your `/usr/local/etc/ncp.cfg`?

feutl, 2019-09-17:
```
cloud:~$ cat /usr/local/etc/ncp.cfg
{
  "nextcloud_version": "16.0.4",
  "php_version": "7.3",
  "release": "buster",
  "release_issue": [
    "Debian GNU/Linux 10",
    "Raspbian GNU/Linux 10"
  ]
}
```
For me - this seems fine ;)

nachoparker, 2019-09-18:
See if this fixes it for you

```
source /usr/local/etc/library.sh
install_app nc-restore
```

and try again

feutl, 2019-09-23:
> 
> 
> See if this fixes it for you
> 
> ```
> source /usr/local/etc/library.sh
> install_app nc-restore
> ```
> 
> and try again

thanks, going to do so, but also will take some time, because I need to do this on the production system and for whatever reason users start complaining if the NC instance is down for a time :D :D

closed 2020-01-19 by nachoparker

feutl, 2020-01-20:
Sorry for never responding. Your advise fixed the problem. Thanks