nextcloud/nextcloudpi #1008: install.sh do not work on buster : check_distro error

erdoukki, 2019-09-30:
the install.sh script no more work (devel and master)...
I fix it by modifying etc/library/sh

`function check_distro()
{
  local cfg="${1:-$NCPCFG}"
  local supported=$(jq -r .release < "$cfg")
  local distro=$(lsb_release -sc)
  if [ $distro == $supported ]; then
    return 0
  else
    return 1
  fi
}

`

erdoukki, 2019-09-30:
without this fix, I get following error (in DEBUG mode)
```
+ check_distro etc/ncp.cfg
+ local cfg=etc/ncp.cfg
++ jq -r .release
+ local supported=buster
+ grep -q buster /dev/fd/63
++ lsb_release -sc
+ return 1
+ echo 'ERROR: distro not supported:'
ERROR: distro not supported:
+ cat /etc/issue
Debian GNU/Linux 10 \n \l

+ exit 1

```

nachoparker, 2019-10-06:
Actually the io redirection is not needed. I don't get why it fails for you but let's just remove it.

closed 2019-10-06 by nachoparker