nextcloud/nextcloudpi #1011: backup abort because of chown

the-matzup, 2019-10-03:
**System information**
`NextCloudPi version  v1.17.1
NextCloudPi image    NextCloudPi_07-21-19
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sdb sdc 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.2G/120G
rootfs usage         2.1G/7.1G
swapfile             /swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    16.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:01`

**Bug-Report**
When I perform a backup with ncp-backup, the process is aborted with the following error message: 
`chown: changing group of '/media/USBdrive/ncp-backups/nextcloud-bkp_20191003_1570136572.tar': Operation not permitted
Abort...`

By deleting the following line in the ncp-backup script, the backup function works fine again (I don't know why the user group has to be changed for the backup):
`chown :www-data "$destfile"`

nachoparker, 2019-10-08:
Thanks for repoting. I don't see why changing the group would not be permitted, as long as it is not a windows filesystem, which is not supported.

closed 2019-10-08 by nachoparker

nocin, 2020-02-24:
I had the same issue doing the backups on an ZFS share I have mounted to my NextcloudPi.
So my situation is a little different, but if anyone ends up here, this was my solution:

The problem is since ZFS on Linux 0.8.3 the no_root_squash is not set as default anymore. All files were created as nobody:nogroup and chown wasn't possible anymore.

Enabling no_root_squash again fixed it:
`zfs set sharenfs='rw,no_root_squash' zpool/data`

As alternative use /etc/exports/
`# sudo nano /etc/exports
/zpool/data/ *(rw,no_subtree_check,sync,insecure,no_root_squash)`

aleistah, 2021-04-05:
> **Bug-Report**
> When I perform a backup with ncp-backup, the process is aborted with the following error message:
> `chown: changing group of [...]

Tried that and found out, that without the chown you get trouble, when restoring.
but u can fix that manually:

1. in case of restoring copy the Backup file to local folder on nextcloudpi
2. "chown :www-data" the backup file before restoring
3. restore the smoothly prepared backup file in nextcloudpi panel

RayCulp, 2021-07-28:
Just in case anyone has this problem with a USB drive with FAT, exFAT or NTFS filesystem on it, you need to format the drive with a Linux filesystem, like ext4. 

1. Find path of USB stick
`lsblk`
[image]

2. If mounted, unmount it
`sudo umount /mnt/usbstick/` (example)

3. Create filesystem
`sudo mkfs.ext4 /dev/sda1` (example)

4. Mount it
`sudo mount -t auto /dev/sda1 /mnt/usbstick/`

KlausHans, 2023-07-24:
Stumbled on this issue also. Thx @RayCulp, exFat was the problem in my case.
After `sudo mkfs.ext4 /dev/sda1` i would suggest an `sudo e2label /dev/sda1 backup` to give the partition a descriptive name for the usb automount function. With the automount function of Nextcloudpi you don't need to fiddle with fstab, it gets mounted on `/media/backup`.