nextcloud/nextcloudpi #1014: Install.sh on debian buster in LXC container

Tjevin, 2019-10-16:
**System information**

Running the install.sh - either using curl or by dwonloading and running locally, even with sudo - yields two issues.

The first issue is failure of the check distro. This is easy circumwented by entirely ignoring that.

The second is when it tryes to setup redis where it fails for starting the service with permission denied.

`Selecting previously unselected package php-igbinary.
(Reading database ... 17508 files and directories currently installed.)
Preparing to unpack .../php-igbinary_3.0.0-1_amd64.deb ...
Unpacking php-igbinary (3.0.0-1) ...
Selecting previously unselected package php-redis.
Preparing to unpack .../php-redis_4.2.0-1_amd64.deb ...
Unpacking php-redis (4.2.0-1) ...
Setting up php-igbinary (3.0.0-1) ...
Setting up php-redis (4.2.0-1) ...
Job for redis-server.service failed.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
`

I have tried running the install.sh script with sudo but alas it makes no difference.

Up until that point in the install process, all seems to go smoothly.

Tjevin, 2019-10-16:
Full transcript:

root@nextcloud:~# apt-get install curl
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  ca-certificates libcurl4 libldap-2.4-2 libldap-common libnghttp2-14 libpsl5 librtmp1 libsasl2-2 libsasl2-modules
  libsasl2-modules-db libssh2-1 openssl publicsuffix
Suggested packages:
  libsasl2-modules-gssapi-mit | libsasl2-modules-gssapi-heimdal libsasl2-modules-ldap libsasl2-modules-otp
  libsasl2-modules-sql
The following NEW packages will be installed:
  ca-certificates curl libcurl4 libldap-2.4-2 libldap-common libnghttp2-14 libpsl5 librtmp1 libsasl2-2 libsasl2-modules
  libsasl2-modules-db libssh2-1 openssl publicsuffix
0 upgraded, 14 newly installed, 0 to remove and 0 not upgraded.
Need to get 2644 kB of archives.
After this operation, 5272 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://ftp.us.debian.org/debian buster/main amd64 openssl amd64 1.1.1c-1 [842 kB]
Get:2 http://ftp.us.debian.org/debian buster/main amd64 ca-certificates all 20190110 [157 kB]
Get:3 http://ftp.us.debian.org/debian buster/main amd64 libsasl2-modules-db amd64 2.1.27+dfsg-1 [69.0 kB]
Get:4 http://ftp.us.debian.org/debian buster/main amd64 libsasl2-2 amd64 2.1.27+dfsg-1 [106 kB]
Get:5 http://ftp.us.debian.org/debian buster/main amd64 libldap-common all 2.4.47+dfsg-3+deb10u1 [89.6 kB]
Get:6 http://ftp.us.debian.org/debian buster/main amd64 libldap-2.4-2 amd64 2.4.47+dfsg-3+deb10u1 [225 kB]
Get:7 http://ftp.us.debian.org/debian buster/main amd64 libnghttp2-14 amd64 1.36.0-2 [84.9 kB]
Get:8 http://ftp.us.debian.org/debian buster/main amd64 libpsl5 amd64 0.20.2-2 [53.7 kB]
Get:9 http://ftp.us.debian.org/debian buster/main amd64 librtmp1 amd64 2.4+20151223.gitfa8646d.1-2 [60.5 kB]
Get:10 http://ftp.us.debian.org/debian buster/main amd64 libssh2-1 amd64 1.8.0-2.1 [140 kB]
Get:11 http://ftp.us.debian.org/debian buster/main amd64 libcurl4 amd64 7.64.0-4 [332 kB]
Get:12 http://ftp.us.debian.org/debian buster/main amd64 curl amd64 7.64.0-4 [264 kB]
Get:13 http://ftp.us.debian.org/debian buster/main amd64 libsasl2-modules amd64 2.1.27+dfsg-1 [104 kB]
Get:14 http://ftp.us.debian.org/debian buster/main amd64 publicsuffix all 20190415.1030-1 [116 kB]
Fetched 2644 kB in 4s (649 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package openssl.
(Reading database ... 10907 files and directories currently installed.)
Preparing to unpack .../00-openssl_1.1.1c-1_amd64.deb ...
Unpacking openssl (1.1.1c-1) ...
Selecting previously unselected package ca-certificates.
Preparing to unpack .../01-ca-certificates_20190110_all.deb ...
Unpacking ca-certificates (20190110) ...
Selecting previously unselected package libsasl2-modules-db:amd64.
Preparing to unpack .../02-libsasl2-modules-db_2.1.27+dfsg-1_amd64.deb ...
Unpacking libsasl2-modules-db:amd64 (2.1.27+dfsg-1) ...
Selecting previously unselected package libsasl2-2:amd64.
Preparing to unpack .../03-libsasl2-2_2.1.27+dfsg-1_amd64.deb ...
Unpacking libsasl2-2:amd64 (2.1.27+dfsg-1) ...
Selecting previously unselected package libldap-common.
Preparing to unpack .../04-libldap-common_2.4.47+dfsg-3+deb10u1_all.deb ...
Unpacking libldap-common (2.4.47+dfsg-3+deb10u1) ...
Selecting previously unselected package libldap-2.4-2:amd64.
Preparing to unpack .../05-libldap-2.4-2_2.4.47+dfsg-3+deb10u1_amd64.deb ...
Unpacking libldap-2.4-2:amd64 (2.4.47+dfsg-3+deb10u1) ...
Selecting previously unselected package libnghttp2-14:amd64.
Preparing to unpack .../06-libnghttp2-14_1.36.0-2_amd64.deb ...
Unpacking libnghttp2-14:amd64 (1.36.0-2) ...
Selecting previously unselected package libpsl5:amd64.
Preparing to unpack .../07-libpsl5_0.20.2-2_amd64.deb ...
Unpacking libpsl5:amd64 (0.20.2-2) ...
Selecting previously unselected package librtmp1:amd64.
Preparing to unpack .../08-librtmp1_2.4+20151223.gitfa8646d.1-2_amd64.deb ...
Unpacking librtmp1:amd64 (2.4+20151223.gitfa8646d.1-2) ...
Selecting previously unselected package libssh2-1:amd64.
Preparing to unpack .../09-libssh2-1_1.8.0-2.1_amd64.deb ...
Unpacking libssh2-1:amd64 (1.8.0-2.1) ...
Selecting previously unselected package libcurl4:amd64.
Preparing to unpack .../10-libcurl4_7.64.0-4_amd64.deb ...
Unpacking libcurl4:amd64 (7.64.0-4) ...
Selecting previously unselected package curl.
[... 1260 lines cut ...]
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.3-redis'
The following additional packages will be installed:
  php-igbinary
The following NEW packages will be installed:
  php-igbinary php-redis
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 196 kB of archives.
After this operation, 817 kB of additional disk space will be used.
Get:1 http://ftp.us.debian.org/debian buster/main amd64 php-igbinary amd64 3.0.0-1 [67.5 kB]
Get:2 http://ftp.us.debian.org/debian buster/main amd64 php-redis amd64 4.2.0-1 [128 kB]
Fetched 196 kB in 1s (159 kB/s)
Selecting previously unselected package php-igbinary.
(Reading database ... 17508 files and directories currently installed.)
Preparing to unpack .../php-igbinary_3.0.0-1_amd64.deb ...
Unpacking php-igbinary (3.0.0-1) ...
Selecting previously unselected package php-redis.
Preparing to unpack .../php-redis_4.2.0-1_amd64.deb ...
Unpacking php-redis (4.2.0-1) ...
Setting up php-igbinary (3.0.0-1) ...
Setting up php-redis (4.2.0-1) ...
Job for redis-server.service failed.
See "systemctl status redis-server.service" and "journalctl -xe" for details.

Tjevin, 2019-10-22:
I dropped the LXC approach so this can be closed.

closed 2019-10-22 by Tjevin

nachoparker, 2021-09-29:
added experimental support for LXC/LXD

https://github.com/nextcloud/nextcloudpi/commit/212f4e01325978b9bd937171814e564164c5f4af

there will be an image next release