nextcloud/nextcloudpi #1018: Cannot start the docker container with bind-mount on windows

szaimen, 2019-10-26:
Hello, thank you for your great work! I love using nextcloudpi.

I am trying to start a docker container with the following command on Windows 10:

`docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v "C:\test":/data  --name nextcloudpi ownyourbits/nextcloudpi-x86`

Docker is correctly installed and the folder on the host "C:\test" gets correctly populated with the new files. But if I click on "activate" in the webpanel, it just says "NextCloudPi not yet initialized, trying again in a few seconds ..." for a view hours now.
So I think this is a bug and I would be really happy, if that could get fixed.
I tried other thinks but this error is ocurring every thime, I try to use the bind-mount.

szaimen, 2019-10-27:
In the logs I get this:
[image]

nachoparker, 2020-01-19:
Sorry but we do not support docker on windows

closed 2020-01-19 by nachoparker