nextcloud/nextcloudpi #1024: 401 for keeweb app within nextcloudpi-docker

stephanfriedrich, 2019-11-09:
ho thx for your awesome nextcloudpi projekt.
i use nextcloudpi-docker an it works like a charme, exept from the keeweb app.

i open the app and want to open a keepassX file via webdav, but get an 401 issue.

maybe its because the apache-server is not well configured within nextcloudpi-docker.

here is the additionall config for keeweb for apache-server, maybe it should addes to your apache congi ?

https://github.com/keeweb/keeweb/wiki/WebDAV-Config

greetings

sunjam, 2021-10-15:
I'm able to recreate this error.  I believe the fault here is with keeweb itself.  [See this list of related issues](https://github.com/keeweb/keeweb/issues?q=is%3Aissue+is%3Aopen+401).  The app works fine when opening from a local directory.

nachoparker, 2021-10-15:
Thanks, we'll tag as external bug. If somebody wants to dig in and finds a solution changing apache config we can include it in NCP