nextcloud/nextcloudpi #1029: Docker run command need update

ZEROF, 2019-11-15:
Hi,

Most of user running last version of Rasbian have this issue, and in my case I have docker on Pi v3.

You need to update your README with this command:

docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi $DOMAIN

to

docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v $(pwd)/nextcloud/ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi $DOMAIN

Best regards ;)

mad73923, 2019-11-21:
It should be clear to the user who runs docker its necessary to enter a valid path.

nachoparker, 2020-01-19:
Thanks for the suggestion, but the README instructions use a volume instead of a host volume. Both are valid options

closed 2020-01-19 by nachoparker