nextcloud/nextcloudpi #1032: Backup of database only

jandamm, 2019-11-23:
**Description**  
I'm using NCP with S3 as primary storage. 
If I'm informed right, when my sd-card breaks or my nc-database is corrupted my files aren't readable at all.
To reduce the risk, I would like to make regular backups of my database.
While this is currently possible as part of the `nc-backup-auto`, the backup files are about 150MB for my setup (which is pretty basic).
The Database would be much smaller. So I'd love to make a regular backup of the database only.

The restoration could be to replace the database file inside the `.tar.gz` file or by a new `nc-restore-db` option.
Then you would have to `nc-restore` and then `nc-restore-db`

Co0p3r21, 2019-12-01:
I think this app could be a suitable solution for you https://apps.nextcloud.com/apps/nextbackup

ovpc, 2019-12-01:
No, even its dev advise using something more robust, it only backups structure, no data.

If I'd really need just db, I'd use mysqldump

`mysqldump -uncadmin nextcloud   > /home/pi/nextcloud.sql`

Run as root or with sudo will create a backup of your db.

theCalcaholic, 2019-12-12:
If you're not using nextcloud's encryption, your files will remain readable even if the database is broken. They will even be available in NC again (after restoring the db) if you run nc-scan.

theCalcaholic, 2019-12-13:
Hm... How exactly did you setup the S3 storage?

jandamm, 2019-12-13:
Like this
```yaml
array (                                   
    'class' => '\\OC\\Files\\ObjectStore\\S3
',                                          
    'arguments' =>                          
    array (                                 
      'bucket' => '1-nextcloud',            
      'autocreate' => true,                 
      'key' => 'KEY',      
      'secret' => 'SECRET',                           
      'hostname' => 's3.eu-central-1.wasabis
ys.com',                                    
      'port' => 443,                        
      'use_ssl' => true,                    
      'region' => 'eu-central-1',           
    ),                                      
  )
```

theCalcaholic, 2019-12-16:
Oh cool. I wasn't aware that NC supports this natively. I have to read through the documentation in order to judge how nc-scan works with this kind of external storage setup.