nextcloud/nextcloudpi #1036: nc-scan does not handle all utf-8 characters in the path

hunhejj, 2019-12-11:
Tried to run nc-scan for a path that contains a special UTF-8 character (`ó`, i.e. `U+00F3`). 
Got a `path not found: ...` error as the letter was simply stripped from the path.

**System information**
ncp 1.19.1

nachoparker, 2019-12-14:
Hi,

I suggest you escalate this bug report to nextcloud/server

Thanks

hunhejj, 2019-12-14:
Don’t you think this is related to the web interface of nextcloudpi? When I reload the page, the path shows the wrong address, i.e. without those characters. Meaning it doesn’t save it correctly either.

nachoparker, 2019-12-19:
uh is this in ncp-web? yes, then it would be a ncp issue

theCalcaholic, 2019-12-19:
Maybe it's related to the issue I discuss in #1037? @nachoparker What do you think about my proposed solution there (only implemented for space characters so far, but could be extended for anything)?