nextcloud/nextcloudpi #1043: Error: Could not download app calendar

jdainen, 2019-12-27:
Error: Could not download app calendar.

Then Nextcloud will not run, but Nextcloudpi will run.

This is on a new install.  If I delete the ncdata folder and reinstall.  It still fails.

Is this a server issue?

Godzgift206, 2020-01-07:
Same issue. I did have it installed working great, but when I installed “letsencrypt” docker things didn’t work as well.

nachoparker, 2020-01-19:
Sometimes the hosting for some apps are shaky and while they are down they cause this failure

closed 2020-01-19 by nachoparker

WillPower3309, 2020-04-16:
This issue still exists as of April 16, 2020

FackJox, 2020-04-21:
I get this same error, wipe ncdata, wipe volumes, wipe images, same compose file but no luck!
> Same issue. I did have it installed working great, but when I installed “letsencrypt” docker things didn’t work as well.
Funnily enough things went bad for me when messing round with the letsencrypt docker too ! It worked for a bit but I restarted the server and then nextcloud stopped working.