nextcloud/nextcloudpi #1045: nc-previews 3 out of 4 processes stop generating previews

sserkez, 2019-12-29:
Hi,
when executed on rock64, 3 out of 4 processes stop generating previews.
terminal output:

2019-12-29T18:37:01+00:00 Scanning folder /<username>/files/Media/Photos/2019
2019-12-29T18:37:13+00:00 Scanning folder /<username>/files/Media/Photos/2019/2019                                     .01_France
2019-12-29T18:40:05+00:00
2019-12-29T18:40:05+00:00 In Folder.php line 162:
2019-12-29T18:40:05+00:00
2019-12-29T18:40:05+00:00   [OCP\Files\NotPermittedException]
2019-12-29T18:40:05+00:00   Could not create folder
2019-12-29T18:40:05+00:00
2019-12-29T18:40:05+00:00
2019-12-29T18:40:05+00:00 Exception trace:
2019-12-29T18:40:05+00:00  () at /var/www/nextcloud/lib/private/Files/Node/Folde                                     r.php:162
2019-12-29T18:40:05+00:00  OC\Files\Node\Folder->newFolder() at /var/www/nextclo                                     ud/lib/private/Files/AppData/AppData.php:155
2019-12-29T18:40:05+00:00  OC\Files\AppData\AppData->newFolder() at /var/www/nex                                     tcloud/lib/private/Preview/Generator.php:419
2019-12-29T18:40:05+00:00  OC\Preview\Generator->getPreviewFolder() at /var/www/                                     nextcloud/lib/private/Preview/Generator.php:115
2019-12-29T18:40:05+00:00  OC\Preview\Generator->getPreview() at /var/www/nextcl                                     oud/lib/private/PreviewManager.php:205
2019-12-29T18:40:05+00:00  OC\PreviewManager->getPreview() at /var/www/nextcloud                                     /apps/previewgenerator/lib/Command/Generate.php:184
2019-12-29T18:40:05+00:00  OCA\PreviewGenerator\Command\Generate->parseFile() at                                      /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:171
2019-12-29T18:40:05+00:00  OCA\PreviewGenerator\Command\Generate->parseFolder()                                      at /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:169
2019-12-29T18:40:05+00:00  OCA\PreviewGenerator\Command\Generate->parseFolder()                                      at /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:145
2019-12-29T18:40:05+00:00  OCA\PreviewGenerator\Command\Generate->generatePathPr                                     eviews() at /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:11                                     5
2019-12-29T18:40:05+00:00  OCA\PreviewGenerator\Command\Generate->execute() at /                                     var/www/nextcloud/3rdparty/symfony/console/Command/Command.php:255
2019-12-29T18:40:05+00:00  Symfony\Component\Console\Command\Command->run() at /                                     var/www/nextcloud/3rdparty/symfony/console/Application.php:901
2019-12-29T18:40:05+00:00  Symfony\Component\Console\Application->doRunCommand()                                      at /var/www/nextcloud/3rdparty/symfony/console/Application.php:262
2019-12-29T18:40:05+00:00  Symfony\Component\Console\Application->doRun() at /va                                     r/www/nextcloud/3rdparty/symfony/console/Application.php:145
2019-12-29T18:40:05+00:00  Symfony\Component\Console\Application->run() at /var/                                     www/nextcloud/lib/private/Console/Application.php:213
2019-12-29T18:40:05+00:00  OC\Console\Application->run() at /var/www/nextcloud/c                                     onsole.php:97
2019-12-29T18:40:05+00:00  require_once() at /var/www/nextcloud/occ:11
2019-12-29T18:40:05+00:00
2019-12-29T18:40:05+00:00 preview:generate-all [-p|--path [PATH]] [--] [<user_id                                     >]
2019-12-29T18:40:05+00:00
2019-12-29T18:40:34+00:00
2019-12-29T18:40:34+00:00 In Folder.php line 162:
2019-12-29T18:40:34+00:00
2019-12-29T18:40:34+00:00   [OCP\Files\NotPermittedException]
2019-12-29T18:40:34+00:00   Could not create folder
2019-12-29T18:40:34+00:00
2019-12-29T18:40:34+00:00
2019-12-29T18:40:34+00:00 Exception trace:
2019-12-29T18:40:34+00:00  () at /var/www/nextcloud/lib/private/Files/Node/Folde                                     r.php:162
2019-12-29T18:40:34+00:00  OC\Files\Node\Folder->newFolder() at /var/www/nextclo                                     ud/lib/private/Files/AppData/AppData.php:155
2019-12-29T18:40:34+00:00  OC\Files\AppData\AppData->newFolder() at /var/www/nex                                     tcloud/lib/private/Preview/Generator.php:419
2019-12-29T18:40:34+00:00  OC\Preview\Generator->getPreviewFolder() at /var/www/                                     nextcloud/lib/private/Preview/Generator.php:115
2019-12-29T18:40:34+00:00  OC\Preview\Generator->getPreview() at /var/www/nextcl                                     oud/lib/private/PreviewManager.php:205
2019-12-29T18:40:34+00:00  OC\PreviewManager->getPreview() at /var/www/nextcloud                                     /apps/previewgenerator/lib/Command/Generate.php:184
2019-12-29T18:40:34+00:00  OCA\PreviewGenerator\Command\Generate->parseFile() at                                      /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:171
2019-12-29T18:40:34+00:00  OCA\PreviewGenerator\Command\Generate->parseFolder()                                      at /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:169
2019-12-29T18:40:34+00:00  OCA\PreviewGenerator\Command\Generate->parseFolder()                                      at /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:145
2019-12-29T18:40:34+00:00  OCA\PreviewGenerator\Command\Generate->generatePathPr                                     eviews() at /var/www/nextcloud/apps/previewgenerator/lib/Command/Generate.php:11                                     5
2019-12-29T18:40:34+00:00  OCA\PreviewGenerator\Command\Generate->execute() at /                                     var/www/nextcloud/3rdparty/symfony/console/Command/Command.php:255
2019-12-29T18:40:34+00:00  Symfony\Component\Console\Command\Command->run() at /                                     var/www/nextcloud/3rdparty/symfony/console/Application.php:901
2019-12-29T18:40:34+00:00  Symfony\Component\Console\Application->doRunCommand()                                      at /var/www/nextcloud/3rdparty/symfony/console/Application.php:262
2019-12-29T18:40:34+00:00  Symfony\Component\Console\Application->doRun() at /va                                     r/www/nextcloud/3rdparty/symfony/console/Application.php:145
2019-12-29T18:40:34+00:00  Symfony\Component\Console\Application->run() at /var/                                     www/nextcloud/lib/private/Console/Application.php:213
2019-12-29T18:40:34+00:00  OC\Console\Application->run() at /var/www/nextcloud/c                                     onsole.php:97
2019-12-29T18:40:34+00:00  require_once() at /var/www/nextcloud/occ:11
2019-12-29T18:40:34+00:00
2019-12-29T18:40:34+00:00 preview:generate-all [-p|--path [PATH]] [--] [<user_id                                     >]

the last 4th process continues to load cpu and apparently yield new previews:
htop -> /bin/bash /usr/local/bin/ncc preview:generate-all -n -v -p /<username>/files/Media/Photos/2019

btw, in ncp-report one should probably consider marking "trusted_domains" also as *sensitive value*

**System information**

<details>
[... 115 lines cut ...]
```

{"reqId":"XgfcSzejx852qgDc@yTbEwAAAA0","level":2,"time":"2019-12-28T22:50:54+00:00","remoteAddr":"2.203.242.150","user":"--","app":"core","method":"GET","url":"\/ocs\/v2.php\/apps\/spreed\/api\/v1\/signaling\/settings","message":"Login failed: 'svitozar' (Remote IP: '2.203.242.150')","userAgent":"Mozilla\/5.0 (Android) Nextcloud-Talk v7.0.8","version":"16.0.4.1"}
{"reqId":"XgfcUDejx852qgDc@yTbFAAAAA8","level":2,"time":"2019-12-28T22:51:04+00:00","remoteAddr":"2.203.242.150","user":"--","app":"core","method":"GET","url":"\/ocs\/v2.php\/cloud\/capabilities","message":"Login failed: 'svitozar' (Remote IP: '2.203.242.150')","userAgent":"Mozilla\/5.0 (Android) Nextcloud-Talk v7.0.8","version":"16.0.4.1"}
{"reqId":"XgiQFCqN63hPs7kDeAyN3wAAAAU","level":2,"time":"2019-12-29T11:37:58+00:00","remoteAddr":"2.203.242.150","user":"--","app":"core","method":"GET","url":"\/ocs\/v2.php\/apps\/spreed\/api\/v1\/signaling\/settings","message":"Login failed: 'svitozar' (Remote IP: '2.203.242.150')","userAgent":"Mozilla\/5.0 (Android) Nextcloud-Talk v7.0.8","version":"16.0.4.1"}
[... cut ...]
{"reqId":"hn8aE8LVIbsx3y2j2ZMq","level":2,"time":"2019-12-29T18:40:05+00:00","remoteAddr":"","user":"--","app":"music","method":"","url":"--","message":"FileHooks::updated triggered for a non-existing file","userAgent":"--","version":"16.0.4.1"}
{"reqId":"jTLouJSOleSGWCiwFz9Q","level":2,"time":"2019-12-29T18:40:10+00:00","remoteAddr":"","user":"--","app":"music","method":"","url":"--","message":"FileHooks::updated triggered for a non-existing file","userAgent":"--","version":"16.0.4.1"}

```
</details>

nachoparker, 2020-01-19:
> btw, in ncp-report one should probably consider marking "trusted_domains" also as sensitive value

very good point. You should report this to nextcloud/server

guandalf, 2022-01-14:
I think this must be fixed in another way now:

`In Application.php line 669:

  [Symfony\Component\Console\Exception\CommandNotFoundException]
  [Symfony\Component\Console\Exception\CommandNotFoundException]
  Command "preview:pre-generate" is not defined.
  Command "preview:pre-generate" is not defined.

  Did you mean one of these?
  Did you mean one of these?
      migrations:generate
      migrations:generate
      notification:generate
      notification:generate
      preview:repair
      preview:repair
`