nextcloud/nextcloudpi #1046: NCP docker setup

goldyka, 2019-12-31:
Hi,

I've been reading for hours and hours on forums... but i still cant fix it.

I set up /etc/fstab
`UUID="6d0ab6c5-cef1-4fc4-9fdb-261cfdc99ce1" /media/ncp ext4 rw,users 0 0`

output of lsblk 
`
sdb           8:16   0 596.2G  0 disk
└─sdb1        8:17   0 596.2G  0 part /media/ncp
`
Cleaned docker, nextcloudpi works if i have everything in the normal location, but as soon as i remove the container, delete it and want to create a new container with the following command:

`docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/ncp/data:/data --name nextcloudpi ownyourbits/nextcloudpi-armhf norbertbiro.ddns.net
`

I get this error code:
`/run-parts.sh: line 42: /etc/services-enabled.d/000ncp: Permission denied
/run-parts.sh: line 47: /etc/services-enabled.d/010lamp: Permission denied
/run-parts.sh: line 47: /etc/services-enabled.d/020nextcloud: Permission denied
Init done
`

Help me please :)

ovpc, 2019-12-31:
Go to forum and post your question there, include output of
>  ls -lh /media/ncp/

Looks like you just need to create
/media/ncp/data
and/or change its ownership

closed 2019-12-31 by ovpc

goldyka, 2019-12-31:
`ls -lha /media/ncp`

total 12K
drwxr-xr-x 3 pi   root 4.0K Dec 31 10:07 .
drwxr-xr-x 3 root root 4.0K Dec 31 07:11 ..
drwxr-xr-x 7 pi   pi   4.0K Dec 31 10:10 data