nextcloud/nextcloudpi #1047: samba release only for one/some nextcloud user

hurrikan77, 2020-01-02:
**Description**  
Actually it is possible to activate the sambe sharing function for all users which are created within the nextcloud instance. I would like to have only sambe release for selected user/users.

**Example**  
Nextcloud User 1
Nextcloud User 2
Nextcloud User 3

After selecting the navigation link I would like to have an selection field for which user/users I would like to aktive the samba sharing function.

theCalcaholic, 2020-01-03:
That would be a nice feature, but is hard to implement with the current ncp-app architecture.
Dynamic configuration items are currently not supported (they are read from static text files atm).

So... good idea, but probably too much effort right now I'd expect.

hurrikan77, 2020-01-03:
If you create for each user an user group "e.g. samba" within nextcloud.
Is there an possibility to add to the loop for creating samba users an request if the user is member of the "samba" User group.
If yes create the samba sharing, if not skip the user.

theCalcaholic, 2020-01-03:
Not a bad idea. A "filter_by_group" option might work

theCalcaholic, 2020-01-04:
@hurrikan77 Out of curiosity: What's your use case for that feature? Why do you want to enable samba only for a portion of your users?

nachoparker, 2020-01-08:
current support for the feature is very basic. All samba clients share the same password...

hurrikan77, 2020-01-08:
@theCalcaholic 
I have done a little bit of rasperry pi "recycling" during the holidays. 
I set up some Volumio instances to play music in different rooms. Base is an music depot which I host in my nextcloud instance. I created an new user for that and shared the files of this user via samba with each instance. So I could prevent to copy the whole data on each of the pi's.

That is the case why I asked for only one/some user. For all other users ist not nessesary to have samba Access to there files.

hurrikan77, 2020-01-08:
@nachoparker 
And thats one reason why I want to reduce the number of open shares to a minimum.

nachoparker, 2020-03-03:
the PR @theCalcaholic implemented is a step forward. Still no password per user or anything like that...

Thanks @theCalcaholic

closed 2020-03-03 by nachoparker