nextcloud/nextcloudpi #1049: NextCloudPi Update not persistant after docker-compose down

paschaef, 2020-01-07:
When using the ownyourbits/nextcloudpi-armhf docker image the NextCloudPi version is at v1.15.2.
After updating NextCloudPi we arrive at version v.1.20.1 which is the latest at the moment.
When restarting the docker container the versions remains at v.1.20.1 as expected.
However, when stopping the container by e.g. docker-compose down and then start it again with docker-compos up the NextCloudPi version is again at v.1.15.2.
The version seems not to persist if the docker container is completely shut down and then started again.

nachoparker, 2020-01-19:
Hi,

That is to be expected, since the image doesn't change and changes to the container get removed when the container disappears. If you don't want that to happen you have to `docker commit` the container and save it as an image.

But there is really no need to destroy the container, you can just stop and start it. Even if you do your NC instance is unaffected.

closed 2020-01-19 by nachoparker

paschaef, 2020-01-20:
Ok, thanks for the hint.
I thought this information would also be stored persistently in the mounted `/data` volume.