nextcloud/nextcloudpi #1052: ncp-backup archive meta information should be cached

theCalcaholic, 2020-01-10:
**Description**  
If the limit for ncp-backup is set to higher values (e.g. 60 in my case), the generation of meta information for all packages (to be shown in the web interface) can take very long (one `tar -I pigz -tf ...` process takes about 5s on my RasPi (for dataless backups) which results in 60 * 5s = 5min delay before the web interface is usable).
This is especially annoying, since no ncp-apps can be run from the web interface during that time.

Instead, it would be nice to cache that information (e.g. in a separate file per archive), so it must only be generated for new backups. Whenever a backup is deleted, the corresponding meta information should also be deleted.

nachoparker, 2020-01-11:
sounds like a good idea

nachoparker, 2020-04-18:
implemented in `v1.24.3`

https://github.com/nextcloud/nextcloudpi/commit/ce9a99fc4d0d935d7c88a9d4670629f05789c96f

Thanks

closed 2020-04-18 by nachoparker