nextcloud/nextcloudpi #1053: Ownyourbits Down

feutl, 2020-01-13:
Domain expired, it happened again https://github.com/nextcloud/nextcloudpi/issues/206

```
Domain Name: OWNYOURBITS. COM
Registry Domain ID: 2089651605_DOMAIN_COM-VRSN
Updated Date: 2020-01-13T08:47:03Z
Creation Date: 2017-01-12T20:22:07Z
Registry Expiry Date: 2021-01-12T20:22:07Z
```

ovpc, 2020-01-14:
Thanks for reporting. fixed now it seems.

closed 2020-01-14 by ovpc

feutl, 2020-01-14:
still down at least for me, so re-open the ticket until more than just one person reports it is fixed.
wohis tells me that the domain was renewed, but this is just not enough ;)

@ovpc  have you tested it ?

ovpc, 2020-01-14:
It has been reported on NCP telegram group to work again by several members, and 
[image]
Loads fine here, DNS check is without errors.
Maybe should clear browser cash, or your DNS is slow and hasn't picked it up yet?

coreybruce, 2022-07-22:
domain is completely down

theCalcaholic, 2022-07-22:
The site is available again.  If it isn't working for you, please verify that the same is true for a different browser (and ideally, device) as well. It could be a caching issue on your client

coreybruce, 2022-07-23:
> The site is available again. If it isn't working for you, please verify that the same is true for a different browser (and ideally, device) as well. It could be a caching issue on your client

Not working for me, I tried it on my phone as well as my PC and it's down. I also tried a is it down website when I made this report and today and it still says it's down, I can't even ping the website..

[image]

[image]
[image]
[image]

What has happened to this project? it use to work than the website went down and it's hard to find the working image for the raspberry pi and I've had issues with the image I did have for the pi 4 breaking

theCalcaholic, 2022-07-23:
@coreybruce Hi, the official page of NextcloudPi is https://nextcloudpi.com. (not linked to ownyourbits.com) It should be available.

In regard to your question: I recently took over project maintainership from Nacho and we started migrating all the hosting related things from his servers to other solutions. In the meanwhile, some things weren't available, but everything should now slowly fall into place.

The official page is https://nextcloudpi.com
The Wiki can be found at https://docs.nextcloudpi.com
Images can be found at https://github.com/nextcloud/nextcloudpi/releases

Sorry for any confusion that was caused by the migration

coreybruce, 2022-07-23:
Ahh right I see, thank you for informing me @theCalcaholic, is there a latest stable version of nextcloud pi or only v1.47.2-rc currently? :smile:

theCalcaholic, 2022-07-23:
> Ahh right I see, thank you for informing me @theCalcaholic, is there a latest stable version of nextcloud pi or only v1.47.2-rc currently @theCalcaholic ? :smile:  

Currently, 1.47.2-rc is the way to go. I'm working on the 1.48.2 image atm. :)

But 1.47.2 is confirmed working and will update to 1.48 with `ncp-update`

coreybruce, 2022-07-24:
Ok thanks,  I'll give it a go :smile: