nextcloud/nextcloudpi #1055: ncp panel to execute shell scripts

burgerkingGH, 2020-01-14:
Would it be possible to use the ncp panel to execute an arbitrary shell script? We would specify recurring (with a frequency of execution), or one-time-only, and the path for the .sh file. 

The reason for this would be to allow very simple testing and integration of arbitrary shell script code with ncp and no external configuration of cron jobs. Specifically, I was wondering about this because I would like to set up a cron job executing duplicity to create encrypted backups.

Thank you for this amazing code that has saved me countless hours over the years.

theCalcaholic, 2020-01-16:
Hm... That's more a question of 'should we' than 'can we'...
My personal opinion (I'm not the maintainer, though) is, that this feature would encourage dangerous behaviour.
Users who want to do such a thing should be versed enough to setup a cron job or execute a script via the terminal (if not, they should probably keep their hands off it :wink: ).

nachoparker, 2020-01-19:
yeah, IMO if a user wants to tweak the system they have all the power of SSH to do so, and this feature could potentially introduce security bugs.

Thanks for the idea though

closed 2020-01-19 by nachoparker