nextcloud/nextcloudpi #1056: nc-update "integer expression expected"

stucksubstitute, 2020-01-16:
installed via curl on Raspberry 3+ on latest Buster lite with all updates
[ncp-report.txt](https://github.com/nextcloud/nextcloudpi/files/4072408/ncp-report.txt)
. Running nc-update throws various error messages. 

Running nc-update
Downloading updates
Performing updates
**grep: /usr/local/etc/ncp-version: No such file or directory**
**/usr/local/etc/library.sh: line 262: [: : integer expression expected**
/usr/local/etc/library.sh: line 264: [: : integer expression expected
/usr/local/etc/library.sh: line 266: [: : integer expression expected
/usr/local/etc/library.sh: line 262: [: : integer expression expected
/usr/local/etc/library.sh: line 264: [: : integer expression expected
/usr/local/etc/library.sh: line 266: [: : integer expression expected
/usr/local/etc/library.sh: line 262: [: : integer expression expected
/usr/local/etc/library.sh: line 264: [: : integer expression expected
/usr/local/etc/library.sh: line 266: [: : integer expression expected
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Fetched 15.0 kB in 2s (6,465 B/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Config value jpeg_quality for app preview set to 60
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
**Failed to enable unit: Unit file haveged.service does not exist.**
Done. Press any key...

**System information**

XDA-Bam, 2020-01-18:
Had the same issue on my Raspberry Pi 4B. After a clean install, the file /usr/local/etc/ncp-version is missing and all ncp updates fail.

As a workaround, you can simply create the file yourself using the appropriate version number. For me, it was 'v1.19.1', but anything older than the most recent release should trigger the update.

nachoparker, 2020-01-19:
fixed, thanks for reporting!

closed 2020-01-19 by nachoparker

devilsharp, 2020-02-17:
Hello,

I have got the same issue

`Failed to enable unit: File haveged.service: No such file or directory`

When I perform nc-update.
I have checked /usr/local/etc/ncp-version and it is present with 'v1.14.3'.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.14.3
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      66G/1,8T
rootfs usage         2,1G/15G
swapfile             /media/b355efe8-a481-4814-a4ed-e96dbae4aba5/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    17.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               13days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.0.44",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "4": "raspicloud.net",
            "11": "2a01:e34:edeb:58e0:998c:3303:314f:cca",
            "12": "192.168.0.8"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "17.0.2.1",
        "overwrite.cli.url": "https:\/\/raspicloud.net\/",
        "overwritehost": "raspicloud.net",
        "overwritewebroot": "\/",
[... 71 lines cut ...]
```

{"reqId":"XkqVfH8AAQEAAG4LDegAAACI","level":2,"time":"2020-02-17T13:30:36+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"XkqVmn8AAQEAADD-NQoAAAAT","level":2,"time":"2020-02-17T13:31:06+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"XkqVm38AAQEAAG4LDekAAACM","level":2,"time":"2020-02-17T13:31:07+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/index.php","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"XkqVuH8AAQEAAG4LDeoAAACO","level":2,"time":"2020-02-17T13:31:36+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"XkqV1n8AAQEAAG4LDesAAACQ","level":2,"time":"2020-02-17T13:32:06+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"XkqV9H8AAQEAAG4LDewAAACJ","level":2,"time":"2020-02-17T13:32:36+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"XkqWEn8AAQEAADD-NQsAAAAG","level":2,"time":"2020-02-17T13:33:06+00:00","remoteAddr":"81.255.54.145","user":"devilsharp","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
[... cut ...] ud\/data\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/80.0.3987.87 Safari\/537.36","version":"17.0.2.1"}
{"reqId":"HvXyT2LabjkBEU9EjKzm","level":2,"time":"2020-02-17T13:37:41+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"\/occ","message":"Temporary directory \/var\/www\/nextcloud\/data\/tmp is not present or writable","userAgent":"--","version":"17.0.2.1"}

```
</details>

devilsharp, 2020-03-03:
Hello @nachoparker,

I am still blocked in v1.14.3, I don't receive any update since this point. Do you want additional info on my config to see where is the issue ?

nachoparker, 2020-03-04:
try  `apt install haveged`

devilsharp, 2020-03-04:
Thank you @nachoparker it works 👍