nextcloud/nextcloudpi #1057: Mail server, fully integrated with Nextcloud

budulinek, 2020-01-17:
Hi,
I have noticed that you already use postfix for sending notification emails ( https://github.com/nextcloud/nextcloudpi/issues/147 ). What about adding full-fledged mail server?

I am a long-time user of Nextcloud (and Owncloud before that). Couple of months ago I have successfully launched an email server (postfix+dovecot+rspamd), FULLY INTEGRATED with Nextcloud. Your email (yourname@yourdomain.com) serves as Nextcloud account. New Nextcloud users automatically get their mailbox. Same login and password for Nextcloud and for email (IMAP and SMTP). You can use webmail in Nextcloud for accessing your mailbox without additional login.

I have followed this excellent tutorial:

https://123qwe.com/

The tutorial is not mine, I am just drawing your attention to it in case you would like to include email server in NextCloudPi. I am on Debian Buster. I have closely followed the tutorial and it works! I would only suggest to use Rainloop app, instead of default Email app. Rainloop has much better UI and does not require additional app "Auto Mail Accounts". As a theme for Rainloop, use https://github.com/LukWebsForge/NextRainCloud .

nachoparker, 2020-01-17:
That looks really nice. I would be happy to include it. Do you feel brave to send a PR?

budulinek, 2020-01-17:
I do not feel brave enough for pull request. I do not have much experience with github, I am not a developer. 

But I will be in touch. 
I have only found NextCloudPi yesterday, so I will have to try it first to see how it works and what is already in. I use LXC (rather than Docker) on my Raspberry, so I will try to build new container for testing NextCloudPi (if possible .... https://github.com/nextcloud/nextcloudpi/issues/1014 https://github.com/nextcloud/nextcloudpi/issues/870 )

First thing I noticed: the tutorial uses PostgreSQL, but you have MariaDB. So I suppose we will have to adjust the SQL commands used for initial creation of databases and tables.  Which is something beyond my skills...