nextcloud/nextcloudpi #1058: LXC

budulinek, 2020-01-18:
OK, so this is my experience with installing NextCloudPi in LXC. I know, it is not officially supported. But you can still run  Curl installer within LXC. There have been previous unsuccessful attempts to install NCP within LXC: https://github.com/nextcloud/nextcloudpi/issues/1014 https://github.com/nextcloud/nextcloudpi/issues/870 

My environment is the "most friendly one": LXC (debian buster) hosted on the Raspberry Pi itself (default armbian with debian buster). Clean LXC.

Curl installer finished without major errors shown. Packages installed, services launched OK, including fail2ban.service (https://github.com/nextcloud/nextcloudpi/issues/870) and redis-server.service (https://github.com/nextcloud/nextcloudpi/issues/1014). Only dnsmasq.service failed to start, because port 53 was already busy. But I do not think this is a big issue. 
I also noticed one error "Failed to connect to codeload.github.com port 443". 

I have relaunched curl installer for the second time, this time no "Failed to connect" error. Process ended OK. /var/www/nextcloud was present, apache running...

But..... No activation page. I have noticed that /var/www/ncp-web is missing.
I have checked the install scripts and noticed bin/ncp-update $BRANCH. So I manually launched `/usr/local/bin/ncp-update master`
And finally, NCP installed. At the end, updater complained that "Nextcloud is not installed - only a limited number of commands are available" (even though Nextcloud was successfully installed by the curl installer) and " There are no commands defined in the "config:app" namespace.". 
So far so good, I finally have the activate page in my browser.

budulinek, 2020-01-18:
NextCloudPi not yet initialized, trying again in a few seconds ...
taking more than just few seconds. Some problems with ncp-launcher:
`[Sat Jan 18 12:01:24.964326 2020] [authz_host:error] [pid 1802:tid 2790319136] [client 192.168.1.194:55752] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name`

budulinek, 2020-01-18:
OK, so I commented out
Require host localhost
in site config and I no longer have the authz_host:error.

But initialization not successful, i have still this errors in ncp.log:
```
[ nc-admin ]
Nextcloud is not installed - only a limited number of commands are available

  There are no commands defined in the "user" namespace.

```

So it seems I have to go back to the installation process.

budulinek, 2020-01-18:
OK, I found it. Installation did not complete. See https://github.com/nextcloud/nextcloudpi/pull/1059

budulinek, 2020-01-18:
OK, I can confirm that NextCloudPi was successfully installed in LXC - at least in my case (LXC hosted by Raspberry Pi itself). Nextcloud itself running.
Of course this is just LXC, so as expected, some things do not work (nc-hdd-test, nc-automount ...).
Issue closed.

closed 2020-01-18 by budulinek

nachoparker, 2020-01-19:
thanks for the heads up. I am happy to accept PRs to improve the installer script so that it runs on lxc.

Best

nachoparker, 2021-09-29:
added experimental support for LXC/LXD

https://github.com/nextcloud/nextcloudpi/commit/212f4e01325978b9bd937171814e564164c5f4af

there will be an image next release