nextcloud/nextcloudpi #1060: Will Nextcloud 18.0.0 be available soon?

maschhoff, 2020-01-18:
Hey Guys,

thanks for building this awesome projekt!
Will Nextcloud 18.0.0 be available soon?

Regards
knex

nachoparker, 2020-01-19:
hi,

you can install it yourself through ncp-update-nextcloud at your own risk. I have recieved reports that it works fine. We will release to autoupdate once we have tested it ourselves.

nachoparker, 2020-01-20:
I tried out the upgrade just now and it works fine, but it disables a bunch of apps that need to be re-enabled in the apps menu because they are not marked as compatible yet with NC18 (but they work).

For this reason, we will wait a bit probably until NC18.0.1 to make sure that major bugs are fixed and apps are up to date before we roll out to all users that use nc-autoupdate-nc to upgrade automatically.

That being said anybody can upgrade manually at their own risk with nc-update-nextcloud

maschhoff, 2020-01-20:
Thanks! I had to mannually enter 18.0.0 as version to make it.

lenzls, 2020-01-26:
I heard in a recent announcement that starting from version 18 nextcloud gets rebranded to "nextcloud hub" and has the onlyoffice integration build in.

Did I got that right?

If yes, will that work on arm systems?

I find it hard to get this information from the official website.

ovpc, 2020-01-26:
Updated my (test) rpi3 to NC 18.0.0 using nc-update-nextcloud without any issue, apart from some apps being disabled as mentioned by Nacho. Don't see any build in OnlyOffice functionality, it still needs OO server on a _64 system and the OO app on NC afaik.

Larsene, 2020-01-28:
Right. Integrating OO in the NCP distribution could be a great thing. Plus, with a Raspberry Pi4 (more ram, 64bits) , this should be enough for small groups

feutl, 2020-01-28:
The OnlyOffice integration has to optione, the one with docker and the new version which is called Documentserver (https://github.com/nextcloud/documentserver_community) which has been partly rewritten in PHP.

The docker OO implementation is also supporting ARM where the documentserver only supports x86 as stated on the gihub page.

> The community document server only supports running on x86-64 Linux servers.

silas229, 2020-01-28:
The Nextcloud Hub is not available for me. 

Neither with ncp-config in the terminal
[image]
nor with the Web-Updater
[image]
Do I have to select a different update channel?

lenzls, 2020-01-28:
@silas229 
> Do I have to select a different update channel?

As said in the second comment in this thread, it is not officially supported by ncp.
You have to update manually at your own risk via `ncp-update-nextcloud`

maschhoff, 2020-02-02:
Will this be supported in the ncp docker container as well. Its not possible for me to update (data directory not empty) data is a volume mounted directory.

maschhoff, 2020-02-02:
Everyone the Document Server is only running on x86-64

feutl, 2020-02-26:
Any status update on an upcoming *official* 18.0.1 NCP release ?

nachoparker, 2020-03-03:
18.0.1 was pushed a few days ago

closed 2020-03-03 by nachoparker