nextcloud/nextcloudpi #1061: Modsecurity prevents update of contacts

mehgrmlhmpf, 2020-01-18:
The default configuration of NCP with modsecurity seems to have an issue when handling vcard files.

Setup: current docker instance of NCP with modsecurity enabled, latest updates

Steps to reproduce:
1.  create new contact
2.  save contact (successful)
3. change contact data
4. request error 500 on saving the contact
5. data is lost after site refresh

Steps 3-5 are successfull with disabled modsecurity.

Solution:
- create exception for addressbooksync

The corresponding logfile/line:
`[Sat Jan 18 17:39:21.463016 2020] [:error] [pid 23201:tid 1817326624] [client xxxxxx] [client xxxxxxx] ModSecurity: XML parser error: XML: Failed parsing document. [hostname "xxxxxxxx"] [uri "/remote.php/dav/addressbooks/users/xxxxxx/xxxxxx/xxxxxx-2DA4-4988-8FC6-xxxxxxxxxx.vcf"] [unique_id "XiNCyZLwCYQ4v2CWduyWOQAAShg"]`

nachoparker, 2020-01-19:
Hi, 

Thanks for reporting. It would be great to tweak modsec config to fix this, please let us know if you find out the fix.

amischler, 2021-11-08:
Hi,

I'm facing the same issue.

As a workaround I added the following line to /etc/modsecurity/modsecurity.conf to skip other rules for the addressbooks requests :

    SecRule REQUEST_FILENAME "@contains /remote.php/dav/addressbooks/" "id:1,ctl:ruleEngine=Off"

I'm not submitting a PR, since there might be a better approach. Skipping all rules might introduce security issues.

Moreover, I'm wondering whether the issue should be fixed in the Contacts app directly. Actually, the Content-Type of the request made by the Contact app is "application/xml" but the body of the request is a VCard, that's why an XML parsing exception is raised by the modsec request body processor. Maybe the Content-Type of the request should be something like "application/vcard".

Any other ideas about this ?

joshtrichards, 2025-02-04:
Related: nextcloud/contacts#196