nextcloud/nextcloudpi #1065: fail2ban not responding to failed logins

stucksubstitute, 2020-01-22:
Similar issue to #116 and #338. fail2ban seems to run but doesn't actually respond to failed login attempts, even though these get registered in nextcloud.log.

Not a great situation because it gives a false sense of security. I hope this is just me and those previous cases reported are really just isolated edge cases.

I tried reverting the timezone to UTC as advised in #116 but that results in an internal server error (I'd have to go back in time from Central European Time, so that makes sense). So that's not an option.

Using the latest Raspbian Buster lite. Installed via the curl installer.  

I'm accessing Nextcloudpi from remote on a different port (if that matters). Something like https://mynamehere.dyndns.com:63234. This gets forwarded by the router to port 443. This has worked fine for years.

The path to nextcloud.log in /etc/fail2ban/jail.conf is correct.   

Also tried. 
```service rsyslog restart```  
```fail2ban stop``` 
```fail2ban start``` 

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```
NextCloudPi version  v1.20.1
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda sdb 
datadir              /media/SSD1/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      52G/224G
rootfs usage         2.1G/14G
swapfile             /var/swap
dbdir                /media/SSD1/ncdatabase_
Nextcloud check      ok
Nextcloud version    17.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               17min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.178.84",
            "2": "xxxxxxxxx:15213",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "11": "2b0a:a540:5764:0:d26f:798:7fe5:11d7"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost\/",
        "dbtype": "mysql",
        "version": "17.0.2.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/media\/SSD1\/ncdata\/nextcloud.log",
        "loglevel": "2",
        "log_type": "file",
        "theme": "",
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "587",
        "mail_smtpauth": 1,
        "mail_smtpsecure": "tls",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "enable_previews": false,
        "tempdirectory": "\/media\/SSD1\/ncdata\/tmp",
        "jpeg_quality": "60"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Jan 22 00:00:11.798687 2020] [mpm_event:notice] [pid 936:tid 1995780624] AH00489: Apache/2.4.38 (Raspbian) OpenSSL/1.1.1d configured -- resuming normal operations
[Wed Jan 22 00:00:11.798810 2020] [core:notice] [pid 936:tid 1995780624] AH00094: Command line: '/usr/sbin/apache2'
[Wed Jan 22 11:38:18.464378 2020] [mpm_event:notice] [pid 936:tid 1995780624] AH00491: caught SIGTERM, shutting down
[Wed Jan 22 10:38:39.043341 2020] [mpm_event:notice] [pid 809:tid 1995498000] AH00489: Apache/2.4.38 (Raspbian) OpenSSL/1.1.1d configured -- resuming normal operations
[Wed Jan 22 10:38:39.051457 2020] [core:notice] [pid 809:tid 1995498000] AH00094: Command line: '/usr/sbin/apache2'
[Wed Jan 22 10:40:02.608955 2020] [mpm_event:notice] [pid 809:tid 1995498000] AH00491: caught SIGTERM, shutting down
[Wed Jan 22 11:40:20.081701 2020] [mpm_event:notice] [pid 811:tid 1995526672] AH00489: Apache/2.4.38 (Raspbian) OpenSSL/1.1.1d configured -- resuming normal operations
[Wed Jan 22 11:40:20.098495 2020] [core:notice] [pid 811:tid 1995526672] AH00094: Command line: '/usr/sbin/apache2'
[Wed Jan 22 12:02:21.825122 2020] [mpm_event:notice] [pid 811:tid 1995526672] AH00491: caught SIGTERM, shutting down
[Wed Jan 22 12:02:41.338853 2020] [mpm_event:notice] [pid 813:tid 1995477520] AH00489: Apache/2.4.38 (Raspbian) OpenSSL/1.1.1d configured -- resuming normal operations
[Wed Jan 22 12:02:41.340813 2020] [core:notice] [pid 813:tid 1995477520] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2020-01-22 12:03:01 0 [Note] InnoDB: Uses event mutexes
2020-01-22 12:03:01 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2020-01-22 12:03:01 0 [Note] InnoDB: Number of pools: 1
2020-01-22 12:03:01 0 [Note] InnoDB: Using generic crc32 instructions
2020-01-22 12:03:01 0 [Note] InnoDB: Initializing buffer pool, total size = 384M, instances = 1, chunk size = 128M
2020-01-22 12:03:01 0 [Note] InnoDB: Completed initialization of buffer pool
2020-01-22 12:03:01 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2020-01-22 12:03:02 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2020-01-22 12:03:02 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2020-01-22 12:03:02 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2020-01-22 12:03:13 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2020-01-22 12:03:13 0 [Note] InnoDB: 10.3.17 started; log sequence number 637203920; transaction id 225681
2020-01-22 12:03:13 0 [Note] InnoDB: Loading buffer pool(s) from /media/SSD1/ncdatabase_/ib_buffer_pool
2020-01-22 12:03:13 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-01-22 12:03:13 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-01-22 12:03:13 0 [Note] Reading of all Master_info entries succeeded
2020-01-22 12:03:13 0 [Note] Added new Master_info '' to hash table
2020-01-22 12:03:13 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.17-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
2020-01-22 12:03:17 0 [Note] InnoDB: Buffer pool(s) load completed at 200122 12:03:17

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"Xigt6tyJVpWmOw47uzOk0QAAEQE","level":2,"time":"2020-01-22T11:11:44+00:00","remoteAddr":"xxx.xxx.220.180","user":"--","app":"no app in context","method":"POST","url":"\/index.php\/login","message":"Login failed: ncp (Remote IP: xxx.xxx.220.180)","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.13; rv:72.0) Gecko\/20100101 Firefox\/72.0","version":"17.0.2.1"}

```
</details>

nachoparker, 2020-01-22:
if it stopped working it must be because the log line changed, and the regex need to be adjusted

ovpc, 2020-01-22:
Can confirm, it no longer works:

> sudo fail2ban-client status nextcloud
Status for the jail: nextcloud
|- Filter
|  |- Currently failed:	0
|  |- Total failed:	0
|  `- File list:	/media/xnbox/ncdata/nextcloud.log
`- Actions
   |- Currently banned:	0
   |- Total banned:	0
   `- Banned IP list:	

with plenty failed login attempts in nc.log

nachoparker, 2020-01-26:
fixed in #1070 thanks!

stucksubstitute, 2020-01-27:
Thanks for looking into this! The regex matches now, but only if I also comment out the 2nd line from the nextcloud.conf:

```
[INCLUDES]
before = common.conf

[Definition]
failregex = Login failed.*Remote IP.*<HOST>
ignoreregex 
```

If I leave it active, the regex doesn't catch. (not sure why or how this works in detail)

But even with this change, the offending IP doesn't really get blocked. iptables is still wide open:

```
iptables -L INPUT -v -n
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
 2585  206K f2b-nextcloud  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 80,443
```

`fail2ban-client status nextcloud` says
```
Status for the jail: nextcloud
|- Filter
|  |- Currently failed:	1
|  |- Total failed:	7
|  `- File list:	/media/SSD1/ncdata/nextcloud.log
`- Actions
   |- Currently banned:	1
   |- Total banned:	1
   `- Banned IP list:	0.0.0.1
```

stucksubstitute, 2020-01-27:
This regex works for me now. It correctly identifies the IP (and not 0.0.0.1 as above), and successfully bans it

```
[INCLUDES]
#before = common.conf

[Definition]
failregex = ^{.*"message":"Login failed: .* \(Remote IP: <HOST>\)".*}$
ignoreregex =
```

GAS85, 2020-01-27:
Did you test example from this https://gist.github.com/GAS85/957e0b1a4f30120225a7be09b173eb24 or from nextcloud pi commit https://github.com/nextcloud/nextcloudpi/pull/1070/commits/980fa970645a02f8336b03aec1732715956d5aa6 ? Because what you provided is from the gist, not from the nextcloudPI. Difference was that IP is not written between ' any more.

stucksubstitute, 2020-01-27:
With your commit [980fa97](https://github.com/nextcloud/nextcloudpi/commit/980fa970645a02f8336b03aec1732715956d5aa6) did catch, but only if I also commented out the 2nd line in the .conf which said `before = common.conf`- not sure why it was there in the first place. 

But even then it wouldn't function. It mis-identified the IP as 0.0.0.1. 

Only with the combination of commenting out that 2nd line and using the regex from your gist does it work for me.

GAS85, 2020-01-27:
My expectation was that this script and regex worked before and what I did is only update log format that was changed. But seems problem is deeper.

Beowulf2, 2020-01-28:
Well, fail2ban doesn't work on me either...
I have tried the following configurations:
`[INCLUDES]
#before = common.conf

[Definition]
failregex = ^{.*"message":"Login failed: .* \(Remote IP: <HOST>\)".*}$
ignoreregex =`
=> Don't work
`[INCLUDES]
before = common.conf

[Definition]
failregex = ^{.*"message":"Login failed: .* \(Remote IP: <HOST>\)".*}$
ignoreregex =`
=> Don't work
I hope you can help me...

Beowulf2, 2020-01-28:
And unfortunately the standard configuration does not work either...

GAS85, 2020-02-03:
This should be tested before:)

Haraade, 2020-02-10:
The new filter is not working.

Info: https://github.com/fail2ban/fail2ban/issues/2485#issuecomment-516400982

This filter seems to work:

[Definition]

_groupsre = (?:(?:,?\s*"\w+":(?:"[^"]+"|\w+))*)
failregex = ^\{%(_groupsre)s,?\s*"remoteAddr":"<HOST>"%(_groupsre)s,?\s*"message":"Login failed:

datepattern = ,?\s*"time"\s*:\s*"%%Y-%%m-%%d[T ]%%H:%%M:%%S(%%z)?"

Tested in Nextcloud 18.

nachoparker, 2020-02-11:
if someone can implement *and test* something that works I will merge it right away

https://github.com/nextcloud/nextcloudpi/pull/1075

nachoparker, 2020-02-12:
#1078

closed 2020-02-12 by nachoparker

stucksubstitute, 2020-11-24:
@nachoparker there's an update on the filter - https://gist.github.com/GAS85/957e0b1a4f30120225a7be09b173eb24 

Just tested it on my Nextcloudpi running NextCloudPi version  v1.31.0 / Nextcloud version    19.0.4.2 - works fine!