nextcloud/nextcloudpi #1066: ncp update from 1.18.0 to 1.19.1 or higher hangs

pilepalle, 2020-01-23:
Hi,
I am trying to update from 1.18.0 to 1.19.1. or higher. Unfortunately the update breaks with the message “/usr/local/etc/library.sh: line 130: echo: write error: No space left on device”

df prints the following:
Filesystem 1K-blocks Used Available Use% Mounted on
udev 953608 0 953608 0% /dev
tmpfs 204488 22312 182176 11% /run
/dev/mmcblk1p1 15158576 3997372 10970572 27% /
tmpfs 1022424 0 1022424 0% /dev/shm
tmpfs 5120 4 5116 1% /run/lock
tmpfs 1022424 0 1022424 0% /sys/fs/cgroup
tmpfs 1022424 0 1022424 0% /tmp
/dev/zram0 49584 48560 0 100% /var/log
/dev/sda1 3906967324 1777331160 2127747560 46% /media/myCloudDrive
tmpfs 204484 0 204484 0% /run/user/1001

What can I delete from the LOGS?
Or can I enlarge the partition?

What can I do to fix the problem?
Unfortunately my knowledge of Linux is not good.

[Log.txt](https://github.com/nextcloud/nextcloudpi/files/4102215/Log.txt)

pilepalle, 2020-01-23:
I have already been asked about my problem but unfortunately no solution.

https://help.nextcloud.com/t/nextcloudpi-odroid-update-from-1-18-0-to-1-19-1-aborts/66466
https://help.nextcloud.com/t/nextcloudpi-update-funktioniert-nicht/68304

ovpc, 2020-01-23:
I see, the german one is of no use here, NCP's main Dev Nacho does not understand it.
Until we are sure it is a bug, I will close here, and continue on forum. My forum name is OliverV ;-)

closed 2020-01-23 by ovpc