nextcloud/nextcloudpi #1068: Nextcloud Talk with TURN server

feutl, 2020-01-24:
It would be a very nice feature to have the option in the ncp-admin backend to enable/install Talk and a TURN server. As far as I have understood, the configuration is pretty straight forward. I installed one on my machine running NCP and it works just fine.

Some machines won't have sufficient resources, but NCP can be installed not just on Raspis only, this should not be a deal breaker. Also, I would assume the latest Raspi and Rock64s have enough power to provide a proper Talk experience.

English
https://help.nextcloud.com/t/howto-setup-nextcloud-talk-with-turn-server/30794

German
https://decatec.de/home-server/nextcloud-talk-mit-eigenem-turn-server-coturn/

feutl, 2020-02-26:
The coturn team could need some help, especially in handliny let's encrypt symlinks as being used by NCP as well.

https://github.com/coturn/coturn/issues/496

Perhaps someone knows how to resolve this. After my turnserver is stable enough I write together the necessary steps and perhaps some day a scripts grows out of it 💃

feutl, 2020-04-01:
Adjusting permissions for the `archive` directory is needed to get the letsencrypt implementation running with coturn and symlinks.

https://github.com/coturn/coturn/issues/496#issuecomment-607133832

stefan123t, 2020-04-19:
@feutl I have successfully configured the post-hook @misi mentioned in coturn/coturn#496 following the description on stack overflow:

How to setup coturn with letsencrypt
https://serverfault.com/questions/849683/how-to-setup-coturn-with-letsencrypt

This approach is IMHO already streamlined enough to put it into a script for ncp-admin as your original intention was.

Thanks for raising this important question on how to make coturn configuration with let's encrypt certificates available via menu to new nextcloudpi users.

feutl, 2020-08-20:
By the way, this is the official documentation how to setup TURN and configure coturn with Nextcloud Talk

https://nextcloud-talk.readthedocs.io/en/latest/TURN/

I adjusted my setup accordingly again, and works flawless. All left is testing it with NC19 :) which I haven't upgraded to, yet.