nextcloud/nextcloudpi #1069: Nextcloud check error

Seferi, 2020-01-25:
Hi, I have a raspberry pi 4 with a NCP image installed and nothing else. All automatic updates are enabled. It’s been working fine until day before yesterday. I’m getting a ‘internal server error’ when I go to the login site. Nextcloud check gives an error (datadir doesn’t exist).
I posted on the forum to no avail, there are others with the same problem without any answer.
I appreciate if someone could help.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":3,"time":"2020-01-25T16:11:59+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#85","userAgent":"--","version":"17.0.2.1"}
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":2,"time":"2020-01-25T16:11:59+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":3,"time":"2020-01-25T16:12:00+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/media\/USBdrive\/ncdata\/news\/config\/config.ini): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#217","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":3,"time":"2020-01-25T16:12:00+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#85","userAgent":"--","version":"17.0.2.1"}
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":2,"time":"2020-01-25T16:12:00+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":3,"time":"2020-01-25T16:12:00+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"mkdir(): Permission denied at \/var\/www\/nextcloud\/apps\/news\/lib\/AppInfo\/Application.php#142","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":3,"time":"2020-01-25T16:12:00+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#85","userAgent":"--","version":"17.0.2.1"}
{"reqId":"3vxHGDrFbJf7PjTlRcE4","level":3,"time":"2020-01-25T16:12:00+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#86","userAgent":"--","version":"17.0.2.1"}
NextCloudPi version  v1.20.1
NextCloudPi image    NextCloudPi_07-21-19
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          none
datadir              /media/USBdrive/ncdata (doesn't exist)
rootfs usage         2.6G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"8mcdSCHKJ95OxIPXcK9N","level":3,"time":"2020-01-25T16:12:04+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#85","userAgent":"--","version":"17.0.2.1"}
{"reqId":"8mcdSCHKJ95OxIPXcK9N","level":2,"time":"2020-01-25T16:12:04+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"8mcdSCHKJ95OxIPXcK9N","level":3,"time":"2020-01-25T16:12:05+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/media\/USBdrive\/ncdata\/news\/config\/config.ini): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#217","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"8mcdSCHKJ95OxIPXcK9N","level":3,"time":"2020-01-25T16:12:05+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#85","userAgent":"--","version":"17.0.2.1"}
{"reqId":"8mcdSCHKJ95OxIPXcK9N","level":2,"time":"2020-01-25T16:12:05+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"17.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 85
{"reqId":"8mcdSCHKJ95OxIPXcK9N","level":3,"time":"2020-01-25T16:12:05+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"mkdir(): Permission denied at \/var\/www\/nextcloud\/apps\/news\/lib\/AppInfo\/Application.php#142","userAgent":"--","version":"17.0.2.1"}
Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

[... 57 lines cut ...]
2020-01-25 16:10:47 0 [Note] InnoDB: 10.3.17 started; log sequence number 1724842916; transaction id 5413912
2020-01-25 16:10:47 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2020-01-25 16:10:47 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-01-25 16:10:47 0 [Note] Recovering after a crash using tc.log
2020-01-25 16:10:47 0 [Note] Starting crash recovery...
2020-01-25 16:10:47 0 [Note] Crash recovery finished.
2020-01-25 16:10:47 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-01-25 16:10:47 0 [Note] Reading of all Master_info entries succeeded
2020-01-25 16:10:47 0 [Note] Added new Master_info '' to hash table
2020-01-25 16:10:47 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.17-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
2020-01-25 16:10:49 0 [Note] InnoDB: Buffer pool(s) load completed at 200125 16:10:49

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/USBdrive/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

ovpc, 2020-01-25:
> I posted on the forum to no avail, there are others with the same problem without any answer.
I appreciate if someone could help.

Link to these forum posts would help, dont know what and where to look now.
This channel is for bug en docs only. 

Looks like you have a faulty drive or added a new one maybe? 
Without using a proper label for the 1st drive, and with auto-mount enabled, that can mess up things.

closed 2020-01-25 by ovpc

Seferi, 2020-01-25:
@ovpc  Sorry about that. I'll continue on the forum..
(Just as info, the drive works fine, I just copied the whole content just in case and did a test, all good. I did have a second usbstick for the database backups from the beginning on, never caused problems before. I took it out to try, it worked for a minute but then again the same error.. I did format and labeled the drives through NCP webui, what do you mean by proper label? Without the automount enabled, data drive won't be mounted at all... )