nextcloud/nextcloudpi #1072: ncp-update fails if NextCloud user 'ncp' is deleted and new distribution is available.

volanin, 2020-01-31:
Near the end of **update.sh** there is this line to generate a notification if a new distribution is available:

`ncc notification:generate "ncp" "New distribution available" -l "${msg}"`

If the NextCloud user 'ncp' is deleted, this command will fail and **ncp-update** will not complete. Since the availability of Debian Buster, my NextCloudPi wasn't updating properly.

ovpc, 2020-02-03:
ncp is also a system user (UID<1000)  created for specific ncp related system tasks, it should not be removed.

closed 2020-02-03 by ovpc

volanin, 2020-02-03:
Well, I was not talking about the Linux user 'ncp'. I have never deleted that one since, as you said, it's a system user.

I was talking about the NextCloud user 'ncp'. Since I created another admin user in the NextCloud webpanel, I deleted the default user 'ncp' there.

This makes 'ncc notification' fail in the update script.

nachoparker, 2020-02-05:
@volanin is right. I added a quick fix

ovpc, 2020-02-05:
@nachoparker thank you :grinning: