nextcloud/nextcloudpi #1074: Restore Backup from another Device

daschmidt1994, 2020-02-01:
**System information**

Yesterday i tried to restore a backup from a raspberry to a ncpi instance that run on a debian homeserver. The problem is, that the old path to the datadir was /media/btrfs/ncdata and the new one is /var/data/nextcloud/data. the restore always restored in the old one. Problem is the ssd is not big enough to take all the data.

ovpc, 2020-02-01:
ncp has no app for that, you need to edit config.php manually. Please use forum or telegram group for help.

daschmidt1994, 2020-02-01:
i tried it but it dowsent work see yesterday on telegram 
when it doesent work its a issu

ovpc, 2020-02-01:
Not sure how much it would take for nc-restore to have an option; where nc-data location can be different from the backup, from which one is restoring.
It would involve several steps. Editing lines in config and possibly running nc-scan to update the db.

Anyone, feeling up to the task, is free do create a pull request to devel branch. ;-)

daschmidt1994, 2020-02-01:
https://github.com/nextcloud/nextcloudpi/issues/852#issue-423467023

Sorry i saw the sam is at pullrequest

closed 2020-02-01 by daschmidt1994