nextcloud/nextcloudpi #1080: NCP updates not being performed, stuck on 1.20.7

XDA-Bam, 2020-02-21:
Hi. I'm currently running NCP 1.20.7 with NC 17.0.3 on a Raspberry Pi 4 with latest Raspbian. I see that there are more recent NCP versions available, but neither the web updater nor the command line version via ncp-config actually perform an update of NCP.

The output of the web updater is as follows:
```
[ nc-update ]
Downloading updates
Performing updates
```
After that, it just gives the green dot and reloads the page. No error messages, everything runs fine. But it doesn't update. Any suggestions?

ovpc, 2020-02-21:
NC18 is not deemed tested  enough by ncp dev team to be pushed to all users, you can use nc-update-nextcloud and enter 18.0.0 to upgrade, if left to zero it will not.

XDA-Bam, 2020-02-21:
I'm not concerned about NC 18. I know this isn't considered stable, yet. I'm only concerned about the NCP-app. The releases list 1.20.8 and 1.20.9. Why are those not "latest"? Am I missing something?

ovpc, 2020-02-21:
They are patches/improvements waiting to be implemented, will be pushed when deemed save for all users, and when our main dev Nacho decides so.

XDA-Bam, 2020-02-21:
OK, thanks. Just a note then: I find it odd, that these are termed "release" when in fact, they're not released.

closed 2020-02-21 by XDA-Bam