nextcloud/nextcloudpi #1087: Does upgrade to buster also upgrade /usr/local/lib/btrfs-snp ?

prontog, 2020-02-24:
Hello again,

Last week I upgraded to *buster* with `ncp-dist-upgrade`. I have edited a few of your cron scripts to also ping my *healthchecks.io* instance. The health-checks related to `/etc/cron.hourly/btrfs-snp ` started failing after the upgrade. It seems that `/usr/local/bin/btrfs-snp` did not update to the latest version that is *buster* compatible (according to [btrfs-snp](https://github.com/nachoparker/btrfs-snp).

Is this because I edited the cron scripts; If this is not the case, then other installation using **nc-snapshot-auto** from *ncp-config* might also have stopped auto snapshotting...

Again thanks for the great work!

Best regards,

Panos

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.20.7
NextCloudPi image    NextCloudPi_01-17-19
distribution         Debian GNU/Linux 10 \n \l
automount            yes
USB devices          sda sdb 
datadir              /media/nextcloud/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      376G/800G
rootfs usage         6.7G/29G
swapfile             /var/swap
dbdir                /media/nextcloud/ncdatabase
Nextcloud check      ok
Nextcloud version    17.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               4:21

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "192.168.1.167",
            "4": "leontiou.hopto.org",
            "11": "77.49.255.42",
            "20": "leontiou.hopto.org",
            "21": "nextcloudpi.local",
            "22": "192.168.1.167"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "17.0.2.1",
        "overwrite.cli.url": "https:\/\/192.168.1.167\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/nextcloud\/ncdata\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "PLAIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/media\/nextcloud\/ncdata\/nextcloud.log",
        "loglevel": "2",
        "log_type": "file",
        "jpeg_quality": "60",
        "app_install_overwrite": [
            "mail"
        ],
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "25",
        "mail_smtpauth": 1,
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "theme": ""
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Feb 24 00:00:03.121335 2020] [ssl:warn] [pid 2650:tid 548492449136] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Feb 24 00:00:03.139515 2020] [mpm_event:notice] [pid 2650:tid 548492449136] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Mon Feb 24 00:00:03.139594 2020] [core:notice] [pid 2650:tid 548492449136] AH00094: Command line: '/usr/sbin/apache2'
[Mon Feb 24 12:56:53.854859 2020] [mpm_event:notice] [pid 2650:tid 548492449136] AH00491: caught SIGTERM, shutting down
[Mon Feb 24 12:57:55.054228 2020] [ssl:warn] [pid 2595:tid 547577288048] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Feb 24 12:57:55.098225 2020] [ssl:warn] [pid 2650:tid 547577288048] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Feb 24 12:57:55.104134 2020] [mpm_event:notice] [pid 2650:tid 547577288048] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Mon Feb 24 12:57:55.104280 2020] [core:notice] [pid 2650:tid 547577288048] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2020-02-24 12:57:47 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2020-02-24 12:57:47 0 [Note] InnoDB: Number of pools: 1
2020-02-24 12:57:47 0 [Note] InnoDB: Using generic crc32 instructions
2020-02-24 12:57:47 0 [Note] InnoDB: Initializing buffer pool, total size = 896M, instances = 1, chunk size = 128M
2020-02-24 12:57:47 0 [Note] InnoDB: Completed initialization of buffer pool
2020-02-24 12:57:47 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2020-02-24 12:57:47 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2020-02-24 12:57:47 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2020-02-24 12:57:47 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2020-02-24 12:57:48 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2020-02-24 12:57:48 0 [Note] InnoDB: Waiting for purge to start
2020-02-24 12:57:48 0 [Note] InnoDB: 10.3.22 started; log sequence number 7427788229; transaction id 168199512
2020-02-24 12:57:48 0 [Note] InnoDB: Loading buffer pool(s) from /media/nextcloud/ncdatabase/ib_buffer_pool
2020-02-24 12:57:48 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-02-24 12:57:48 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-02-24 12:57:48 0 [Note] Reading of all Master_info entries succeeded
2020-02-24 12:57:48 0 [Note] Added new Master_info '' to hash table
2020-02-24 12:57:48 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.22-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2020-02-24 12:57:49 0 [Note] InnoDB: Buffer pool(s) load completed at 200224 12:57:49

```
</details>

<details>
<summary>Nextcloud logs</summary>

```
```
</details>

prontog, 2020-03-06:
It seems I did not phrase it correctly. It broke the auto snapshot mechanism on my installation. This is extremely serious by the way. If I did not have monitoring of cron jobs I would never have noticed this. I manually updated /usr/local/bin/btrfs-sn from its repo. The reason I framed it as a question is because it needs checking in other installations (upgraded to buster).

ovpc, 2020-03-06:
Ok, thanks for reporting, checked right away, on two of mine, didn't break nc-snapshot-auto, all snapshots are present as expected, and all are upgraded to Buster. 
So unless and untill anyone can reproduce this, you can post to forum, if still need or want support.
Manually updating is not needed nor recommended, ncp-update takes care of that.

prontog, 2020-03-06:
Thanks @ovpc, I will post to forum. By the way, as I mentioned earlier, I used `ncp-dist-upgrade` to upgrade.

closed 2020-03-06 by prontog

nachoparker, 2020-03-10:
Sounds like we should have upgraded `btrfs-snp` specifically with `ncp-dist-upgrade`. Sorry about that

prontog, 2020-03-11:
Hello @nachoparker, it might be that the `for` loop at [update.sh](https://github.com/nextcloud/nextcloudpi/blob/master/update.sh) does not update already installed entries. I think it already finds the **cfg** files and does nothing (line 85). It seems to me that they should always be installed.

zedocrob, 2022-03-25:
Looks like nc-snapshot-auto is working on my Odroid HC2 (clean flash from latest image).
i can see the snaphots created via SSH but nothing appear on the WebUi.
tried to reactivate, reboot but no luck
[image]
.